PX4/PX4-Autopilot #7679: MISSION_ITEM_REACHED is not sent

imrhch, 2017-07-25:
When mission point is reached, Mission::set_mission_item_reached() is called from Mission::on_active(), setting mission_result.reached to true.
But later in the Mission::on_active(),the set_mission_items() (https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L195) is called and the "reached" flag is reset to false here
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1364 (called from https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L799)

This happens before the "reached" flag is checked in the MavlinkMissionManager::send() and since the mission_result.reached is false (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L350) the message is not sent

I have a feeling that firmware version from last year didn't have this problem.

LorenzMeier, 2017-07-29:
I'm getting this here in HITL successfully. Please re-open if I missed something.

[image]

closed 2017-07-29 by LorenzMeier

imrhch, 2017-07-31:
I'm sorry for inacurate report. I was bit confused that MISSION_ITEM_REACHED is sent when the whole mission is done (I expected it to be sent in every mission point), which may be expected feature. And particularly I missed the MISSION_CURRENT which works perfectly O:)
So thank you very much for directing me.