PX4/PX4-Autopilot #7680: multiple px4 sitl in gazebo

RCDajay, 2017-07-25:
is it possible to simulate multiple drones in gazebo with different px4 sitl, and at the same be controlled using mavros?

TobiMiller, 2017-08-01:
Please follow this https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

RCYajad, 2017-08-08:
I've tried following the article, but i've encountered an error.
[image]

KhristopherK, 2017-08-11:
I also get the same error. They way I got around it was to open that file and run the launch file script in different steps.

EdXian, 2017-08-11:
@KhristopherK  I got the same error too.  Do you mean that i have to do the following steps just like [tutorial ](https://dev.px4.io/en/simulation/ros_interface.html) had said:
1. make xxxx gazebo
2. source file
3. export path
4. roslaunch
 Thanks!!

EdXian, 2017-08-12:
@TobiMiller I have done all steps in [https://dev.px4.io/en/simulation/ros_interface.html](url) and [https://dev.px4.io/en/simulation/multi-vehicle-simulation.html](url) and got the same error . It seems that the file `multi_uav_mavros_sitl.launch` cannot work!!  Do you have a solution for it?

EdXian, 2017-08-13:
@RCYajad  HI , you can try  modify `posix_sitl.launch` which is in `launch` folder like this : 
```
<launch>

    <!-- Posix SITL environment launch script -->
  <group ns="uav1">
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="est" default="lpe"/>
    <arg name="vehicle" default="iris"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/iris.sdf"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)_1"/>
    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
    </node>
     <!-- $(anon vehicle_spawn) -->
    
    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg vehicle)_1 -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
  </group>

  <group ns="uav2">
    <arg name="x" default="1"/>
    <arg name="y" default="1"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="est" default="lpe"/>
    <arg name="vehicle" default="iris"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/iris.sdf"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)_2"/>
    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
    </node>
    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg vehicle)_2 -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
  </group>
 
    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>
    <arg name="ns" default="/"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>

     <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)" />
        <arg name="debug" value="$(arg debug)" />
        <arg name="verbose" value="$(arg verbose)" />
        <arg name="paused" value="$(arg paused)" />
    </include>
</launch>

<!-- vim: set et ft=xml fenc=utf-8 ff=unix sts=0 sw=4 ts=4 : -->

```
Next step you can run `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:145xx" `
to connect each of drones in gazebo. (Notice that before connecting the drone , you have to check its udp port !!) The setting of udp is in files `iris_x` which are in `Firmware/posix-configs/SITL/init/lpe`. 
Good luck!!

RCYajad, 2017-08-13:
Hey, sorry for the late reply but I'm still having trouble in configuring the UDP ports of my drones. And I cant seem to find the iris_x file that you mentioned above.

EdXian, 2017-08-13:
@RCYajad Hi , you can find iris_1 and iris_2 in `Firmware/posix-configs/SITL/init/lpe` 
[image]
and udp port setting.
[image]

If the way I said can work , please tell me !! 
Thanks

EdXian, 2017-08-14:
@RCYajad , You can get more imformation in [here](https://github.com/PX4/Firmware/issues/6408) .

david-spc, 2017-09-09:
Hi, @RCYajad ! I met the same problem as yours, like this:
[image]
Do you have the approach to solve this? thank you

david-spc, 2017-09-09:
Hi, @KhristopherK , I met the same problem as yours, Do you have the approach to solve this now ?

YuYangguang, 2017-10-13:
When I launch the posix_stil.launch , it seems that the px4 node didnot receive data from gazebo, because it always reports like this:
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

Therefore, when I launch the mavros, i can not receive anydata. Did you solve this issue?  @EdXian

EdXian, 2017-10-13:
HI @YuYangguang ,
Did you launch posix_stil.launch correctly?  I think you should set UDP port that is link to gazebo simulator for drone correctly!  You can see [here](https://dev.px4.io/en/simulation/) .
Good Luck!

YuYangguang, 2017-10-13:
I set the udp port in the file: " models/iris.sdf", in which there is a line: <mavlink_udp_port>14560</mavlink_udp_port>.   
Besids, I also set the file iris in the folder "Firmware/posix-configs/SITL/init/lpe"
add this line:param set SITL_UDP_PRT 14560. But it didn't work. It is really weired.
 By the way: which version of firmware are you using?  Thanks.

EdXian, 2017-10-13:
Hi @YuYangguang ,
hmm... it's so strange . 
Maybe you can try to remove your `px4 firmware` and rebuild it. 
```
cd ~/src/Firmware
make posix gazebo
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
```

YuYangguang, 2017-10-14:
I have solve this issue by update my firmware to the latest version. Now i can connect my px4 and mavros. Thanks for your kind  suggestions @EdXian .
But another problem occus. When I tried to control the quad in offboard mode, I can arm the vehicle and chang the mode into offboard. But when I tried to make the quad fly by publishing the data on topics :/mavros/setpoint_position/local. The quad is just still. Nothing happened . Then I ran:"rosrun rqt_graph rqt_graph. It looks like this:
[image]
It likes that the gazebo didn't connect to the px4. But I can get the position and other datas via mavros.
Who can tell me what happened?

EdXian, 2017-10-14:
hi @YuYangguang ,
Can you please paste your code here?

YuYangguang, 2017-10-15:
Thank you for your kind suggestion. @EdXian Today, I update my cmake to v3.9 and update my firmware to the latest version. Everything goes fine. Maybe yesterday I didn't build the firmware well owing to my old cmake version. 

Another thing I want to mention here is that if anyone want to change the Px4 ID in the rcs file , like this:

[image]

please don't forget to modity the arg tgt_system  for the mavros package. like this:

[image]

closed 2018-02-21 by PX4BuildBot