PX4/PX4-Autopilot #7696: Pixhawk 2.1 miscellaneous issues

dagar, 2017-07-28:
List of Pixhawk 2.1 issues. https://github.com/PX4/Firmware/pull/7618

 - I2C driver issues. I2C2 is the px4fmu-v2_default onboard bus. This causes problems for certain i2c drivers expecting I2C magnetometers to be external, or not even trying onboard.
   - mag not recognized as external (https://github.com/PX4/Firmware/issues/7599) *potentially serious*
   - sf1xx failing to start https://github.com/PX4/Firmware/issues/6358#issuecomment-318778733
 - secondary power brick not fully supported. I think this is in place of servo valid

davids5, 2017-07-30:
@dagar Once https://github.com/PX4/Firmware/pull/7618 comes in I will address theses. 

The secondary power will be interesting, in that we will have to move from compile time to run time bick support.

dagar, 2017-09-24:
@davids5 we should take another look at this now that we can detect the version. Users trying to connect I2C devices to the Cube secondary I2C bus (still PX4_I2C_BUS_ONBOARD) is a common pain point.

@proficnc FYI

mrpollo, 2017-10-18:
@bkueng @LorenzMeier please comment

LorenzMeier, 2017-10-31:
I'm looking into this but I will only be able to spend significant time on this tonight.

LorenzMeier, 2017-11-21:
Closed by #8300.

closed 2017-11-21 by LorenzMeier