PX4/PX4-Autopilot #7709: VTOL standard pusher throttle rampup

dagar, 2017-07-31:
VTOL standard transition from MC to FW ramps up the pusher throttle over VT_F_TRANS_DUR seconds.

This doesn't really match the param description and I'm not sure that it's needed at all.

dagar, 2017-08-01:
I think we should drop VT_F_TRANS_DUR from standard and ramp up to VT_TRANS_THR over 1 second.

https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/standard.cpp#L253

@sanderux thoughts?

sanderux, 2017-08-01:
Some props stall at slow speed high throttle. For me it would be fine but there are vehicles that need 4 or 5

dagar, 2017-08-01:
I'm mainly opposed to it being particularly badly named for this usage in VTOL standard.

sanderux, 2017-08-01:
Yes, vt_f_thr_ramp would be better

bresch, 2017-08-02:
Changing the name and description is a good idea. I remember having to look at the code to understand this parameter.

tubeme, 2017-08-05:
We insisted having VT_TRANS_MIN_TM long time ago, to make sure we dont loose the plane in a gusty weather and faulty airspeed readings. Recently (3 months) we started flying airspeedless and at first we thought (as written in the docs) that the transition will take VT_F_TRANS_DUR seconds. But instead it took VT_TRANS_MIN_TM seconds in airspeedless mode. I think VT_F_TRANS_DUR could be put in to use in this case when we have airspeedless transition.

sanderux, 2017-08-05:
@tubeme the docs (https://docs.px4.io/en/advanced_config/vtol_without_airspeed_sensor.html) say nothing about VT_F_TRANS_DUR. that has been the ramp up time for years. It's just a bit poorly named

tubeme, 2017-08-06:
Sorry I meant VT_F_TR_OL_TM... But I percieve it as the VT_F_TRANS_DUR 

From the Parameter refference:

Duration of a front transition

Comment: Time in seconds used for a transition

Something has to change... :)

sanderux, 2017-08-07:
VT_F_TR_OL_TM is currently only used by tilt-rotor. we should probably merge those params.

AndreasAntener, 2017-08-18:
Initially the idea was:
- TRANS_DUR: ramp time, param should be renamed
- MIN_TM: minimum time until we start looking at airspeed (safety measure), also used to starting point in time to weigh down MC outputs when in no airspeed mode
- OL_TM: open loop time, basically the complete duration of a transition used when no airspeed is available. The idea was to introduce that for all types.

Are we so far away from that now?

sanderux, 2017-08-18:
no, that should be a small fix, but will become a breaking change once implemented for existing vtols

tubeme, 2017-08-21:
@AndreasAntener It sounds good to me. @sanderux Right now it is kind of misleading in the Parameters Reference page. What do you mean by braking change? That once flashing the new version the existing VTOL's will break or that the code will brake if the changes are made?

sanderux, 2017-08-22:
@tubeme breaking change means it will break functionality when people upgrade

dagar, 2017-08-22:
If we set the default values conservatively I don't think it's going to be a breaking change. The parameter description is so misleading right now it's unlikely many are using it intentionally.

sanderux, 2017-08-24:
for airspeedsensorless it is a breaking change as the min_tm will have been set (quite high). it is something you would have to tune according to my guide on that. but i'm not sure many people are using airspeedsensorless at the moment. at minimum it should be a release note and maybe even a warning if min_tm > ol_time or if it deviates from the standard

closed 2018-02-21 by PX4BuildBot