PX4/PX4-Autopilot #7719: suddenly drone was fall ...

seungri1, 2017-08-02:
log link = http://review.px4.io/plot_app?log=f0f038ec-0121-4470-9552-73ccf539cb8f
video = https://youtu.be/Sg3jnlwGXhc

i don't know what's problem.  please give me a tip.

LorenzMeier, 2017-08-02:
This is including the hard fault logging but there is no hard fault log on the system, so my first guess would be a power loss, not a software failure. The video audio shows that all propellers stopped during the fall - it becomes completely quiet.

How many flights did this system have before the crash? Did you analyse the electrical connections after the crash?

seungri1, 2017-08-02:
thank you for your advice. let's check it out right away.

I tried it twice and did it twice.

seungri1, 2017-08-03:
you were right!! the problem is solved. thank you!!

mhkabir, 2017-08-03:
Happy flying!

closed 2017-08-03 by mhkabir