PX4/PX4-Autopilot #7723: LPE runaway Z velocity

Seanmatthews, 2017-08-02:
During rest, I'm finding that local_position_estimator, whether immediately or after a small amount of time, starts drifting negative in a linear manner. It's fusing only vis_pos and vis_yaw. The vision pose estimate Z remains at the origin while this is happening.

[image]
* vision_pose_estimate.z = red
* local_position_ned.z = blue
* local_position_ned.vz = light green
* altitude.altitude_local = white

This is just a guess, but it seems like this could be caused by bad values in the [noise density value](https://github.com/PX4/Firmware/blob/v1.6.4/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L782)? Where do those values come from? My IMU data looks alright, with no jumps at rest.

Details
* Snapdragon Flight (software v3.1.3.1)
* PX4 v1.6.4

Seanmatthews, 2017-08-02:
Sometimes good measurements last several minutes before the problem begins:
[image]

Seanmatthews, 2017-08-02:
Does it make sense that estimated Z velocity hovers around -0.5?
[image]

TSC21, 2017-08-03:
@Seanmatthews can you please share your LPE parameter list config? Are you fusing baro?

jgoppert, 2017-08-03:
Check your z accel calibration, also I typically have baro fusion on as well.

Seanmatthews, 2017-08-03:
LPE_FUSION is 12 (vis_pos and vis_yaw only). My z_acc is a steady -9.8ish.

TSC21, 2017-08-04:
@Seanmatthews please share the full list of the LPE parameters so to check how they are configured right now.

Seanmatthews, 2017-08-04:
LPE_FUSION       12
LPE_VIS_DELAY  0
LPE_VIS_XY        0.001
LPE_VIS_Z          0.001

TSC21, 2017-08-07:
@Seanmatthews did you try to recalibrate your accel?

kechaofan, 2017-08-08:
I suggest you using INAV estimator instead of LPE. Kalman Filter do have problems like this.

TSC21, 2017-08-08:
> I suggest you using INAV estimator instead of LPE. Kalman Filter do have problems like this.

It has nothing to do with the fact that this is a Kalman Filter. Besides, INAV is becoming deprecated, so not the best of suggestions.

Seanmatthews, 2017-08-08:
@TSC21 I did a recalibration that somewhat ameliorated the issue, though vz still decreased, albeit slowly. After re-flashing the board, the issue seems to have gone away for now. Re-flashing was done with the board hanging sideways, if that matters at all to some initial calibration.

TSC21, 2017-08-08:
> board hanging sideways

What do you mean with "hanging sideways"?

jgoppert, 2017-08-08:
@Seanmatthews you have LPE_VIS_XY outside of the recommended range, that will cause numerical issues with the filter. I would recommend leaving LPE_VIS_XY at 0.1, then you can try LPE_VIS_XY at 0.01, LPE_VIS_XY = 0.001, definitely won't work.

mhkabir, 2017-08-08:
I also remember that there were some issues (in the past; not sure if we fixed it) with accel bias estimation with baro fusion turned off.

EdXian, 2017-08-10:
Hi @Seanmatthews  ,
I have the same problem as yours !!  Have you resolved the problem?

Seanmatthews, 2017-08-15:
@EdXian It's nothing I did in a scientific way. I re-flashed the Snapdragon Flight board while it hung sideways off of my desk by the USB cable. The Z velocity still drifts a small bit, but not by that amount anymore. All my calibration attempts resulted in worse calibrations (incorrect resting X and Y accelerations) than the starting calibration, FYI.

EdXian, 2017-08-15:
@Seanmatthews It seems that the parameter `LPE_VIC_P` should not be set nearly zero.   I'm not sure . I change it to default value and the problem was resolved. How did you set   `LPE_VIC_P` ?

Seanmatthews, 2017-08-15:
@r1b4z01d suggested that this could have something to do with "wind" drafting over the on-board barometer during calibration. I did have my AC on in a small room while flashing and/or calibrating.

mhkabir, 2017-08-15:
No, there is no way for the baro to affect any calibration.

Seanmatthews, 2017-08-15:
If that's the case, then perhaps it affected the barometer while sitting on my desk or floor in front of the AC?

ecmnet, 2017-08-17:
Maybe related to https://github.com/PX4/Firmware/issues/7083

Seanmatthews, 2017-08-24:
@ecmnet I suppose it's possible. There are certain issues with out-of-the-box PX4 that I've noticed and haven't had time to properly look into. For now, I re-calibrated a few times, and then always check the barometer and z values before flight. However, this is still very much an open issue.

mindThomas, 2017-09-06:
@mhkabir 
> I also remember that there were some issues (in the past; not sure if we fixed it) with accel bias estimation with baro fusion turned off.

I am quite sure this issue is still present. At least we can't seem to get reliable z-performance without enabling the barometer. However by just enabling the barometer and setting the LPE_BAR_Z parameter high seems to do the work.

TSC21, 2017-09-06:
@mindThomas this is a pinned issue that requires resolution. Unfortunately, none of us had the required time to look into it and solve it.

closed 2018-02-21 by PX4BuildBot