PX4/PX4-Autopilot #7727: can not change auto.loiter from offboard use mavros

feilongfl, 2017-08-03:
i use offboard mode to ctrl from ros in gazebo,but i find that i can't exit offboard mode to auto.loiter,it turn to auto.rtl without warning,when it land,it turn to auto.loiter mode.

i use ros kinetic in ubuntu 16.04,and gazebo is in docker use dockerfile `https://github.com/feilongfl/px4-dev-simulation`,the code is master branch.

i post the same issues in `https://github.com/mavlink/mavros/issues/752#issuecomment-319895462`, because i don't know where the error.

closed 2018-02-21 by PX4BuildBot

xuming0617, 2019-01-13:
can anyone tell me ,how can i fix this problem

feilongfl, 2019-01-14:
@xuming0617 Sorry, because this question is too long, I can't remember exactly how to deal with it.
If I remember correctly, we read out the optical flow sensor and ultrasonic data and re-implemented a pos hold function.