PX4/PX4-Autopilot #7729: **Feature Request**  "Wind limit exceeded" warning message

Antiheavy, 2017-08-03:
It would nice to have a parameter for wind speed warning threshold and have a warning message sent down to QGC if the wind estimate exceeds this value during flight.  Defaulted to OFF, but could be settable for air frame specific wind limits.

lakshaydang, 2017-09-26:
is this warning is present on quadcopter? i want to try it on my setup.

dagar, 2017-09-26:
The wind estimate is streamed over mavlink. Handling this on the ground side would probably be better. Then QGC could have an appropriate on screen warning message. The firmware is limited to sending short text messages.

lakshaydang, 2017-09-26:
i was trying to detect the high wind and then change the  mode to RTL on high wind
Is that possible?

dagar, 2017-09-26:
Yes, that type of thing would make sense firmware side. One way to do it would be monitoring the wind_estimate uORB topic in commander and trigger the mode change if it exceeds your limit for a given time period.

lakshaydang, 2017-09-26:
Honestly, i have an idea how to that but when i start implementing it, i don't know how to start. If you can help me with that it would be great. Atleast i would know the path i have to move forward on.
Or if possible can we come on skype?

almaaro, 2019-01-20:
How often the warning should be issued?
- Only once per flight
- Should there also be a warning when the wind speed has dropped to an acceptable level
    - Maybe the wind speed could be estimated with a 2-minute (or so) moving average and issue a warning whenever the limit is exceeded/when the wind speed has dropped

Any ideas?

Antiheavy, 2019-01-20:
> How often the warning should be issued?

good question.  The wind estimate is already filtered, but for a warning like this it probably needs to be more heavily filtered.  1 minute moving average might be fine.  The warning should be able to announce more than once per flight.  Maybe 1 minute moving average, requiring at least 1 minute of data.  Resets upon issuing a warning, causing a minimum of 1 minute again before the next warning.  I'm sure there are lots of ways to implement something similar.

A warning about dropping below an acceptable level is not needed as the warning should prompt the pilot to start watching the wind estimate closely and make decisions about returning to land or changing altitude or something.

bresch, 2022-08-15:
done in https://github.com/PX4/PX4-Autopilot/pull/18343

closed 2022-08-15 by bresch