PX4/PX4-Autopilot #7750: Dealing with SDCARD failure

sanderux, 2017-08-08:
Some flights on a vehicle with a failing SDcard from a client:

Mission read failure (presumably) resulted in unexpected back transition: https://logs.px4.io/plot_app?log=ffa0687e-c817-4163-b52e-4f9fa1a66994 (the user then activated the takeoff waypoint to re-transition)

Complete weirdness at takeoff: https://logs.px4.io/plot_app?log=bc6215da-fd24-425c-a4e8-ecfaa9b12571
Also not all params seem to have been logged

Total failure halfway: https://logs.px4.io/plot_app?log=2258f3eb-fa11-4571-a409-b1d7417f19e3
Also not all params seem to have been logged, this was the output mavlink:

[image]

At minimum,  when mission read fails it must RTL, not loiter around the last known WP
The other strange behavior i am pretty sure is sd card related too, maybe some io buffer issue. 

@LorenzMeier @dagar @bkueng thoughts?

sanderux, 2017-08-08:
PS: on the last log, by the clients account it was in RETURN flight mode but kept loitering around the last waypoint.

dagar, 2017-08-09:
TODO: 
 - investigate existing dataman data integrity checks and possible improvements
 - add failsafe behavior for dataman failure and thoroughly test navigator

sanderux, 2017-08-09:
Also i would do a dry run reading all items before switching. Could we do some crc checking?

LorenzMeier, 2017-08-16:
Yes we should. Sorry for not raising this earlier.

closed 2018-02-21 by PX4BuildBot

JohnSnowball, 2019-05-24:
hello, I see this issue is closed, but I encountered the same problem when I'm using 1.8.2 stable(modified rtk gps driver) firmware. The SD card is broken and then the plane flew to the last known point until the battery ran out. I want to know if you have plans to fix this?