PX4/PX4-Autopilot #7752: Feature request: Turn-Rate Control (Flight Mode)

bgat, 2017-08-09:
Some gliding vehicles, particularly steerable parachutes, offer only yaw and pitch control for steering, rather than the roll and pitch provided by fixed-wing aircraft. Furthermore, the response of a parafoil to braking (yaw) inputs is highly nonlinear with altitude and airspeed, and generally under-actuated. As a result, steering a parachute according to target roll and pitch angles doesn't really work.

A simpler solution is to ignore roll and pitch angles altogether, and just actuate to a target turn rate. An outer loop must then "plan ahead" to request a turn at a rate that can be supported by the vehicle, that also intercepts the waypoint.

There are two use cases where fixed-wing aircraft can also benefit from actively-controlled turn rates: the dreaded, "two-minute hold" loiter, and the maneuvering seen in controlled approaches for landing. In both of these cases the heading change is rate-controlled, but the radius of the resulting turn depends upon the airspeed of the vehicle.

FEATURE DESCRIPTION: a flight mode where the strategy is "do whatever actuator output is necessary to maintain a N-degree/sec heading change".

Note that this feature is NOT Acro mode, because roll and pitch are determined by the requested heading change rate; in Acro mode, roll and pitch rates ARE the targets.

Respectfully submitted,

Bill Gatliff
bgat@billgatliff.com

bgat, 2017-08-09:
Bonus points for "achieve a heading of M-deg at a rate of N-deg/sec".

LorenzMeier, 2017-08-09:
@bgat We're implementing right now "task flight modes" which is essentially an abstract interface which let's you run an instance of a custom class completely take over control, while still taking care of all the system complexity in entering and exiting the mode. So this is entirely possible - what is the timeline you'd like to see that in?

bgat, 2017-08-09:
My test vehicle won't be ready for a month, but probably will want to do some test flying shortly after that. In the meantime, if the opportunity arises I have a Flyzone Calypso on hot-standby. :-)

Where can I track the progress of your development?

Finally, I'm assuming that what you're talking about delivering near-term is the task flight-mode capability, but I'd need to provide the custom class that uses it to do controlled-rate turns?

closed 2018-02-21 by PX4BuildBot