PX4/PX4-Autopilot #7753: Cannot build SITL on Bash on Windows 10

aditya00j, 2017-08-09:
Hi,
I am trying to build the PX4 firmware with Bash on Windows following the instructions from the following webpage:
https://github.com/Microsoft/AirSim/blob/master/docs/px4.md

After installing the tool-chain, I am encountering an error on the following command:
`make posix_sitl_default`

Here is the error:
```
CMake Error at CMakeLists.txt:447 (add_subdirectory):
  add_subdirectory given source "src/lib/DriverFramework/framework" which is
  not an existing directory.

CMake Error at CMakeLists.txt:447 (add_subdirectory):
  The source directory

    /mnt/d/Aditya/Dropbox/GA1/Code/Firmware/src/lib/ecl

  does not contain a CMakeLists.txt file.
```
And then there is a bunch of errors after this, which I am guessing are downstream errors. This happens with the clone from the current master as well as from the v1.6.5 tag. I can see that the folders "ecl" and "DriverFramework"  exist in other branches but haven't been merged to master. Is this error related to this or am I doing something wrong?

I am using the following configuration:
Windows 10 x64 (Build 15063)
Bash on Windows with Ubuntu 16.04
cmake 3.5.2

---

A potentially relevant issue: I was unable to run the following command -
`sudo add-apt-repository ppa:george-edison55/cmake-3.x -y`
I kept getting the error
`gpgkeys: key B118CD3C377DF930EDD06C67084ECFC5828AB726 can't be retrieved`
I can manually retrieve that key but for some reason add-apt-repository won't retrieve it (I am not behind a proxy, and just to be sure, I ran it with the -E flag as well). But since a simple "apt-get install cmake" installs v3.5.2 on Linux 16.04, I thought I should be able to build the firmware.

Thanks for anyone who can help me resolve the issue!

LorenzMeier, 2017-08-09:
It looks like you don't have git installed and / or the git submodules do not get resolved. Bash on Windows is not officially supported for SITL right now and if you really want to push this through you should expect some work to get there. Otherwise we recommend a native Linux installation, Mac OS or a Docker container / VMWare / Virtualbox for SITL.

aditya00j, 2017-08-10:
Thanks for your reply!
I do have git installed on Bash on Windows, but you are right, somehow the packages are not getting pulled using the Makefile (even though the commands for it exist).

I'll try to get it to work, and will post here any useful advice I may find in the process. But perhaps Virtualbox is the way to go for me.

aditya00j, 2017-08-20:
Closing this issue. I couldn't find the problem with git on Bash on Windows, but ended up deciding it was better to move to Linux anyways, especially in order to use the IDE for development.

closed 2017-08-20 by aditya00j

nwallis, 2019-11-22:
Make sure you use
`git clone --recursive`
When cloning your repository