PX4/PX4-Autopilot #7755: vmount causes system freeze

ndepal, 2017-08-09:
I'm trying to use `vmount` with mavlink inputs (`MNT_MODE_IN 3`, i.e. `MAVLINK_DO_MOUNT`) and with mavlink output (`MNT_MODE_OUT 1`, i.e. `MAV_CMD_DO_MOUNT_CONTROL`).

When I configure a PixRacer this way, `vmount` and in fact the whole system freezes: console via debug port no longer responsive, QGC loses connection, PixRacer stops flashing.

### Steps to reproduce
- Flash current master
- reset all parameters
- on the console:
```
param set MNT_MODE_IN 3
param set MNT_MODE_OUT 1
vmount start
```

I suspect the problem lies in the fact that in this configuration, the `vmount` input `InputMavlinkCmdMount` subscribes to `vehicle_command` and the output `OutputMavlink` publishes on it, causing a loop.

I tried preventing `InputMavlinkCmdMount` from processing messages that were published by `OutputMavlink` by checking [here](https://github.com/PX4/Firmware/blob/925efe990d0b17e2c976f51e56860e38b9cc2ba6/src/drivers/vmount/input_mavlink.cpp#L229) that the system and component IDs don't match the one's that `vmount` itself is publishing, but it didn't help.

@bkueng , any ideas?

bkueng, 2017-08-10:
Yep, this is easily reproducible thanks to your precise steps. I'll look into it.

LorenzMeier, 2017-08-13:
Fixed.

closed 2017-08-13 by LorenzMeier