PX4/PX4-Autopilot #7758: FMUv1 Deprecation Notice and removal from source tree.

davids5, 2017-08-09:
PR to follow, that will remove support for FMUv1 HW from PX4

dagar, 2017-08-09:
Remove from QGC flash as well? @DonLakeFlyer FYI

DonLakeFlyer, 2017-08-09:
Or just keep the last stable hanging around and change all versions to that?

dagar, 2017-08-09:
In this cause I don't think the px4fmu-v1 has been functional for at least months. So unless someone wants to go back and find the last acceptable version I think we'll need to drop it completely.

Going forward that would be a good way to deprecate though.

DonLakeFlyer, 2017-08-09:
Ok, I'll pull it.

davids5, 2017-08-09:
Here is the plan: 

I will do the PR removing the FMUv1 from the PX4 code base ASAP.

The PR will state that it is the notice of the FMUv1 Deprecation and will be merged in 30 days. 

From the documentation side.  2 PR should be done:

1) ASAP: Add an announcement of the end of support in 30 day of FMUv1 where FMUv1 is referenced on px4.io  and in readme.
  
2) BY SEP-9 the: Removal of FMUv1 where FMUv1 is referenced on px4.io and in readme.

@hamishwillee Please let me know if you re the correct person to do the documentation PRs?

hamishwillee, 2017-08-09:
> @hamishwillee Please let me know if you re the correct person to do the documentation PRs?

Anyone can do changes to the docs, but yes, I am happy to take care of this. 

> ASAP: Add an announcement of the end of support in 30 day of FMUv1 where FMUv1 is referenced on px4.io and in readme.

Is there a need for a blog post or official PX4 tweet on this? I'm guessing not.

hamishwillee, 2017-08-09:
Often this is mentioned as "PX4FMU-v1+PX4FMU" Do I need to concern myself with mentions of PX4IO?

E.g. https://docs.px4.io/en/advanced_config/parameter_reference.html#BAT_V_SCALE_IO

davids5, 2017-08-10:
do you mean "PX4FMU-v1+PX4IO" ?

hamishwillee, 2017-08-10:
No, I mean PX4IO "stand alone" (I would deal with "PX4FMU-v1+PX4IO"). I assume not - PX4IO support is completely separate from FMU support"

hamishwillee, 2017-08-10:
In [STM32 Bootloader > Run the JLink GDB server](https://dev.px4.io/en/software_update/stm32_bootloader.html#run-the-jlink-gdb-server) it says:

> FMUv1:
```bash
JLinkGDBServer -select USB=0 -device STM32F405RG -if SWD-DP -speed 20000
```
Would this be "Any FMU version" - Ie could I change `FMUv1` to `FMUvX` or should I delete that option?

{the reason I ask is because this is the ONLY FMU version specified}

davids5, 2017-08-10:
> FMUv1
 JLinkGDBServer -select USB=0 -device STM32F405RG -if SWD-DP -speed 20000

Would become 
```
FMUv2

JLinkGDBServer -select USB=0 -device STM32F427VI -if SWD-DP -speed 20000
```
I think the 
```
AeroFC:

JLinkGDBServer -select USB=0 -device STM32F429AI -if SWD-DP -speed 20000
```
Is wrong - the CPU listed in the build is a [CONFIG_ARCH_CHIP_STM32F427V](https://github.com/PX4/Firmware/blob/master/nuttx-configs/aerofc-v1/nsh/defconfig#L262)

It may be more helpful to give a list of targets FMUv2,FMUv3,FMUv4... and the SoC that follow the word
 `-device`

hamishwillee, 2017-08-15:
FYI, https://github.com/PX4/px4_user_guide/pull/48 will finish all user guide work for this. I hope to get it reviewed soon.

davids5, 2017-09-15:
Done in #7769

closed 2017-09-15 by davids5