PX4/PX4-Autopilot #7760: motor_test not part of the nsh commands ?

gjkunde, 2017-08-09:
Dear All,

I selected a 6 rotor airframe and downloaded the firmware for
pixhawk 2.1  but I seem to not see motor_test when I am logged into nsh. 

What could be the reason and how would I test the wiring without the motor_test
command ?

LorenzMeier, 2017-08-09:
Most people just arm the airframe and hold a strip of paper against the motor to see which way it flips (and hence check if rotation is correct). In terms of wiring: Using the wiring diagrams this is usually easy to get right on first try.

I've added `motor_test` now to the FMUv3 target: `make px4fmu-v3_default upload`.

closed 2017-08-09 by LorenzMeier

gjkunde, 2017-08-16:
There seems to be a problem with that, I will open a new issue for this is closed ?