PX4/PX4-Autopilot #7765: Problem with update firmware pixhawk on aero drone

Aurusd, 2017-08-10:
Use this link to update firmware https://dev.px4.io/en/setup/building_px4.html
but have next problem on step 
cd Firmware
make px4fmu-v2_default

Firmware/build_px4fmu-v5_default: No such file or directory make: *** [Makefile:165: px4fmu-v5_default]

what i'am do wrong?

Aurusd, 2017-08-10:
also try on linux.. but result isnt successful.
arthur@arthur:~/dev/Firmware$ sudo make px4fmu-v5_default
-- CMAKE_MODULE_PATH: /home/arthur/dev/Firmware/cmake
-- Nuttx build for px4fmu-v5 on m7 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5-491-g6e35cb0
-- CONFIG: nuttx-px4fmu-v5-default
CMake Error at cmake/toolchains/Toolchain-arm-none-eabi.cmake:29 (message):
  could not find arm-none-eabi-gcc compiler
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:276 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
/bin/sh: 1: cd: can't cd to /home/arthur/dev/Firmware/build_px4fmu-v5_default
Makefile:165: recipe for target 'px4fmu-v5_default' failed
make: * [px4fmu-v5_default] Error 2

Aurusd, 2017-08-10:
actually i want to install pixhawk firmware on aero drone, but result with "make ..." is same

TSC21, 2017-08-10:
`make aerofc-v1_default upload`? Where are you building the code? What's your setup?

Aurusd, 2017-08-10:
how i can make aerofc-v1_default upload if i dont have appropriate *.px4 file?. Both try on linux and win

TSC21, 2017-08-10:
What's the output of `make aerofc-v1_default`?

Aurusd, 2017-08-10:
my setup is like here https://dev.px4.io/en/setup/building_px4.html

Aurusd, 2017-08-10:
i didnt use make aerofc-v1_default.. as i correctly understand i can build any firmware version from PX4 firmware sources

TSC21, 2017-08-10:
https://dev.px4.io/en/flight_controller/intel_aero.html

TSC21, 2017-08-10:
> i didnt use make aerofc-v1_default.. as i correctly understand i can build any firmware version from PX4 firmware sources

That assumption is wrong. Follow what I sent in the previous comment.

Aurusd, 2017-08-10:
you mean i cant flash on aero drone PX4 firmware? Why i ask you, cause i have program code which work fine with PX4, and for i want to make aero uav work with my code. But Mavlink message from flight controller arent same as in PX4. So i am search what is wrong, and my assumption is to flash PX4 firware

Aurusd, 2017-08-10:
All this steps 
Do a full update of all software on the Aero (https://github.com/intel-aero/meta-intel-aero/wiki/Upgrade-To-Latest-Software-Release)
Grab the Firmware
Compile with make aerofc-v1_default
Configure the target hostname
i can make directly on drone via ssh or only on host PC?

Aurusd, 2017-08-10:
i dont think so that next commands will work on drone yocto linux or i am wrong?
Grab the Firmware  its mean -> git clone https://github.com/PX4/Firmware.git
Compile with make aerofc-v1_default

TSC21, 2017-08-10:
> you mean i cant flash on aero drone PX4 firmware? Why i ask you, cause i have program code which work fine with PX4, and for i want to make aero uav work with my code. But Mavlink message from flight controller arent same as in PX4. So i am searching what is wrong, and my assumption is to flash PX4 firware

You can do any modifications on the Firmware that may fit your needs. The reason you build `aerofc-v1_default` is related to board specifications in terms of hardware and the low-level connection to it. The same happens with the other builds, as you can read on the wiki instructions you were following. The builds are specific to each board.

TSC21, 2017-08-10:
> i dont think so that next commands will work on drone yocto linux or i am wrong?
Grab the Firmware its mean -> git clone https://github.com/PX4/Firmware.git
Compile with make aerofc-v1_default

I advise you to completly and calmly follow what is presented to you on the wiki guide. It is explicit enough in my opinion.

Aurusd, 2017-08-10:
ok.. i will try tomorrow

TSC21, 2017-08-10:
Also you should do `git clone https://github.com/PX4/Firmware.git --recursive` to automatically update and clone the submodules.

Aurusd, 2017-08-10:
Also you should do git clone https://github.com/PX4/Firmware.git --recursive to automatically update and clone the submodules. - this do on yocto?

TSC21, 2017-08-10:
>  - this do on yocto?

No. This is for the Firmware. For updating the OS you should follow https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup#flashing-intel-aero-linux-distribution

TSC21, 2017-08-10:
You have everything you need here: https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup

Aurusd, 2017-08-10:
if i already flash latest OS image:intel-aero-image-intel-aero it means i update firmware for flight controller and FPGA too?

Aurusd, 2017-08-10:
you dont know why flight controller may send message (byte[]) not match to mavlink protocol?

Aurusd, 2017-08-11:
make aerofc-v1_default - have next errors
[log.txt](https://github.com/PX4/Firmware/files/1217356/log.txt)

Aurusd, 2017-08-11:
log on ubuntu
[log_ubuntu.txt](https://github.com/PX4/Firmware/files/1217505/log_ubuntu.txt)

Aurusd, 2017-08-14:
> What's the output of make aerofc-v1_default?
> 
make aerofc-v1_default-- CMAKE_MODULE_PATH: /media/ubuntu/MEDIA/Ubuntu/Firmware/cmake
-- Nuttx build for aerofc-v1 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5-518-g891b2f6
-- CONFIG: nuttx-aerofc-v1-default
CMake Error at cmake/toolchains/Toolchain-arm-none-eabi.cmake:29 (message):
  could not find arm-none-eabi-gcc compiler
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:276 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
/bin/sh: 1: cd: can't cd to /media/ubuntu/MEDIA/Ubuntu/Firmware/build_aerofc-v1_default
Makefile:165: recipe for target 'aerofc-v1_default' failed
make: *** [aerofc-v1_default] Error 2

Aurusd, 2017-08-14:
i have done install aerofc-v1_default from intel iso, but still cant read message from mavlink, have crc errors. I have a code which work on PX4, and i dont know is a problem with flight controller firmware or intel aero mavlink has differs from PX4 mavlink

TSC21, 2017-08-14:
You did not install the Toolchain in order to be able to compile/build the Firmware code. https://dev.px4.io/en/setup/dev_env_linux.html

Aurusd, 2017-08-14:
ok.. thanks i will try tomorrow

LorenzMeier, 2017-08-20:
Closing this as stale.

closed 2017-08-20 by LorenzMeier