PX4/PX4-Autopilot #7768: Land detector issues

sanderux, 2017-08-10:
Log: https://logs.px4.io/plot_app?log=e31c9ecf-b862-43df-a419-23efa0aa2ed8
Video: https://drive.google.com/open?id=0B3otHVwcj_LdQ0RSbjFPOU5XaDg

Land detector on VTOL in MC mode does not seem to function properly
The mag timeout and gps loss are indicative of a similar issue to https://github.com/PX4/Firmware/issues/7746

The flight was done by doing a TAKEOFF and then a LAND command.
Similar behavior was in mission mode

Flight was yesterdays master merged with https://github.com/PX4/Firmware/pull/7666

sanderux, 2017-08-10:
@Stifael it seems both maybe_landed and ground_contact are constantly switching as if they are impacting each other. I think the log rate is too low and it is not showing the actual rate of switching the values.

You do see at a certain point many topics are no longer publishing, maybe due to the amount of updates.

After +- 1.5 secs the quad motors even go to 900 while 1025 should be their armed minimum.

sanderux, 2017-08-10:
maybe related? https://github.com/PX4/Firmware/issues/7677

Stifael, 2017-08-11:
@sanderux, that is expected that maybe_landed and ground_contact depend on each other. Please have a look in the devguide [here](https://dev.px4.io/en/tutorials/land_detector.html).

From the log I  saw that MPC_THR_MIN was set to 0.35, which is quite large (for normal quads, we have it set around 0.12). All Landdetection states depend on this value, where ground_contact checks a low thrust criteria for detection [here](https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L381) and maybe_landed and landed  a minimum thrust criteria [here](https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L390). In the case of MPC_THR_MIN = 0.35, the low thrust criteria is satisfied when the thrust setpoint is below 0.395, which is just too large. The thrust setpoints falls below this threshold even  if you just descend. 

I remember we had a similar discussion in the past, but I can't remember your reasoning for setting the thrust min that large.

sanderux, 2017-08-11:
@Stifael we need a high min thurst because a vtol can become unstable during back transition when wind creates lift under the wing. So i would have 2 questions;
1. Do you think this behavior is gone when i set a lower MPC_THR_MIN ?
2. If so, do you think the the current behavior is what the vehicle *should* do when a higher MPC_THR_MIN is set?

I am fairly sure the strange behavior stems from a overload on the uorb as attested to by multiple other topics stopping to publish (vehicle_status and vehicle_attitude for instance). Also, during the problem scenario the thrust setpoint was 0 as the vehicle was considered landed (briefly)

sanderux, 2017-08-11:
One thing i forgot to mention: This behavior was commanded from the GCS, the transmitter was off so the throttle minimum checks were always true.

However, i am not reporting an issue of a false positive, the detection was fine until actual landing was detected, then the system went crazy

Stifael, 2017-08-11:
> Do you think this behavior is gone when i set a lower MPC_THR_MIN ?

Not sure, and probably not (nonetheless, we need to figure out something such that this value can be kept low for vtol as well, but different topic).

It is strange that it detects landing as expected (goes trough the cycle ground_contact, maybe_landed, and landed), but then exit landed and jumps directly into ground_contact, and gets stuck in maybe_landed, although it seems that all criteria are met. In fact, in order to enter landed after maybe_landed, the only difference is time ([here](https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L309)), so I don't really understand why it gets stuck after the second maybe_landed detection

sanderux, 2017-08-11:
I think at that point there is an overload on something, orb or param_write
I think the log rate is too low to see how fast it jumps and that what you see will just be a snapshot of where it was at the log cycle. It's probably switching all of them at a very fast rate

Stifael, 2017-08-11:
I think it show the correct frequency, because in order to reach any landed state, the criteria have to be true for some specific time. It can exit a state immediately, but not enter it.

sanderux, 2017-08-11:
thats true for the landed state, but is that also true for the ground_contact and maybe_landed state?

Stifael, 2017-08-11:
yep, for all of them: https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/LandDetector.h#L149-L156

closed 2017-10-05 by dagar