PX4/PX4-Autopilot #7770: MAVLink/uORB debug messages

jlecoeur, 2017-08-11:
**Feature Request**

- **Issue:**

Using custom MAVLink messages is not convenient. Although generating MAVLink headers is very easy, making new messages accessible in PX4 and QGC is not straightforward.
The first option is to maintain forks of mavlink/c_library_v2,  PX4/Firmware and QGroundcontrol with the new messages and build all from source, which is not ideal in the long term. 
The other option is to have the custom messages integrated in the common MAVLink message set. However, a- not every custom messages can or should make it to common.xml, b- it takes long time to get something accepted in MAVLink, c- it does not fit well with development phase where custom messages are subject to modifications.

- **Proposed solution:**

The goal is to provide a solution for agile development with MAVLink in PX4 and QGC that allows to stay on master, that does not require MAVLink PRs, and that supports quick iteration on message format.
The solution is to use existing MAVLink debug messages, add a new one with larger payload (DEBUG_ARRAY), and make the debug messages transparent between MAVLink network and uORB bus.

Example scenario: An embedded device/sensor needs to exchange data with PX4 in a format that does not fit existing mavlink messages. Data is packed into debug messages with custom format, the debug message is identified by its name field and/or ID field. On PX4 side, the mavlink debug message is converted to debug uORB message and is accessible by the rest of the stack. The other way around, a PX4 task can send custom mavlink message to the device by publishing a debug uORB message. In addition, debug topics can be relayed to QGC and displayed in the analyze view.

- **Status:**

I have 3 PRs that implement the solution above. They are functional but stuck on review (which demonstrates point b- above :-) ):
- https://github.com/mavlink/mavlink/pull/734 adds DEBUG_ARRAY to MAVLink, this is a larger version of DEBUG_VECT for cases that require larger payload. This PR blocks the QGC PR below.
- https://github.com/mavlink/qgroundcontrol/pull/5268 adds support for DEBUG_ARRAY in QGC. Other debug messages are already supported by QGC.
- https://github.com/PX4/Firmware/pull/7373 Adds conversion to/from MAVLink and uORB for debug messages + example app. Support for DEBUG_ARRAY is not included so it can merged independantly of the two first PRs. But if MAVLink PR goes first I will add it.

FYI @LorenzMeier @DonLakeFlyer @TSC21 
I understand it is not easy for you guys to synchronize between 3PRs, can we check here that everything is ready and move it forward?

TSC21, 2017-08-13:
@jlecoeur I perfectly understand and indentify with your problems, as they also happen to me :P though I don't have write permissions on any of the repos, I left my reviews on both mavlink and PX4 repos and hope their are useful and can a trigger so this can be merged soon. @LorenzMeier maybe you can give a hand in this?

jlecoeur, 2018-02-21:
I think that this is still relevant so I reopen.

Status:
- The PX4 PR was accepted.
- The QGC PR was closed, waiting for activity on the MAVLink PR.
- The blocking point is now the MAVLink PR

TSC21, 2018-02-21:
@LorenzMeier can we unstuck this?

jlecoeur, 2018-09-27:
Status:
- the MAVlink PR was merged
- a new PX4 PR was opened: https://github.com/PX4/Firmware/pull/10585
- a new QGC PR was opened: https://github.com/mavlink/qgroundcontrol/pull/6895

jlecoeur, 2018-09-28:
All merged, thanks!

@TSC21 do you want to keep this issue open until there is support in MAVROS?

TSC21, 2018-12-10:
All good! Can be closed

closed 2018-12-10 by TSC21