PX4/PX4-Autopilot #7775: Implement A-GPS via GCS

LorenzMeier, 2017-08-12:
QGroundControl knows its position to a reasonable degree thanks to assisted GPS. We should leverage this to seed the autopilot position.

TSC21, 2017-08-13:
@LorenzMeier so in this case it would be the ground station, which has network connection, that retrieves the satellite data and sends it to the FCU in order to reduce the TTFF. Would it make sense to the FCU have that capability through a GSM/GPRS modem (plus this module could be used to other purposes)?

whyscience, 2017-08-24:
QGC has access to yhe network is easier than adding a GSM/GPRS modem

TSC21, 2017-08-24:
The GSM module could have other purposes, not just dedicated to AGPS

whyscience, 2017-08-24:
Both are needed, A-GPS via GCS is easyier and more useful.

VineeshKK, 2017-09-15:
Is there any open SUPL server exist?
Other option is to get assisting data through GPS vendor portal using a private token. The user will have to register for getting this token. Whether this method of implementation is fine?

TSC21, 2018-02-21:
Seems something worth to discuss and implement.

closed 2019-02-13 by stale[bot]