PX4/PX4-Autopilot #7778: CQWG:instrumentation

davids5, 2017-08-12:
1. Parameter writer lock.
2. FMU RC - Real time Requirement on V4
3. SPI, I2C Jitter
4. Interrupt off time   
5. What are the ramification MS5611 is using HPWORK on SPI  
6. FD left free
7. HPWORK 
 a. What is running?
 b. what are the Runtimes (percounter).
    Look at dispatch - HW Probes - capture longest.  Then add threshold and snapp addresses
8. hrt dispatch - HW Probes - capture longest.  Then add threshold and snapp addresses

davids5, 2017-08-12:
Are IOCTl all thread safe?

dagar, 2017-08-13:
Possibly broader scope, but should we consider bringing the work_queue within PX4? Then we could easily add instrumentation (some on by default) and build it into a base class that eliminates most of the boilerplate for current users.

closed 2018-02-21 by PX4BuildBot