PX4/PX4-Autopilot #7779: PWM channels treated as groups

garfieldG, 2017-08-13:
Looking at fmu driver a little closer it seems like channels are treated as groups inside set_pwm_rate.
https://github.com/PX4/Firmware/blob/861879c6d31eb448a7d427955a14e86e01afc0ff/src/drivers/px4fmu/fmu.cpp#L799

dagar, 2017-08-13:
Yes, and on the nsh console if you run `pwm info` you'll see more information about this. This is due to the timer allocation. Have you encountered a problem with this?

For more information than you probably wanted see this issue. https://github.com/PX4/Firmware/issues/7261

garfieldG, 2017-08-20:
Thanks for response!
I just wanted to point what I meant, inside the method we we iterate over all of the groups and for every group we get the channels assigned to it.
https://github.com/PX4/Firmware/blob/b2a81ed35beb6ab56a5c8d6aaf872e8cc869fd55/src/drivers/px4fmu/fmu.cpp#L822
It doesn't seem right that we iterate over the channels and not over the groups, we will just get zero for most of the iterations.

closed 2018-02-21 by PX4BuildBot