PX4/PX4-Autopilot #7780: FastRTPS: Example from http://dev.px4.io stalls after around one second

LorenzMeier, 2017-08-13:
@VicenteEprosima I've tried to run the example code on my machine this weekend and it did work, but it showed the exit statistics (which are conditioned on `!running`) after about one second.

Is this intended or an issue in the example code? I was expecting that I could consume the data indefinitely.

@mcharleb @bkueng Have you tested the current state on master?

TSC21, 2017-08-14:
@LorenzMeier may you are missing `micrortps_client start -l -1`?

VicenteEprosima, 2017-08-21:
@LorenzMeier The default value are 10000 loops of execution if you want to launch for ever is needed to add the parameter "-l -1" as is pointing by TSC21 above. Do you think this must be the default behavior?

LorenzMeier, 2017-08-21:
Is there any valid use case where you would want this behaviour? It seems so uncommon that I considered this to be a bug.

VicenteEprosima, 2017-08-21:
Ok, @LorenzMeier we changed defaults #7835