PX4/PX4-Autopilot #7784: Gazebo FW SITL doesn't take off properly with certain initial yaws

anassinator, 2017-08-14:
**Bug Report**

I'm running latest master and trying to test a mission in Gazebo SITL but want my plane to start facing the takeoff waypoint. 
I've been able to set the initial yaw successfully with the `Y` ROS launch parameter, but I realized that this doesn't take off properly with certain initial yaws.

To, reproduce, just run:

```bash
roslaunch px4 posix_sitl.launch vehicle:=plane est:=ekf2 headless:=true gui:=false Y:=<YAW>
```

and start the following mission with QGC [yaw_test.plan](https://github.com/PX4/Firmware/files/1222139/yaw_test.plan.txt)

**Test results**

| Initial yaw (rad) | Successful takeoff? | Log |
| --------------- | -------------------- | --- |
| 0.0 (default)  | yes | https://logs.px4.io/plot_app?log=302c2584-c193-4f5d-ad56-b5d4b29e5069 |
| 1.0 | no | https://logs.px4.io/plot_app?log=c6b58cab-776a-4afc-8d3a-43e9255126f3 |
| 2.09 | yes | https://logs.px4.io/plot_app?log=7b1d3c01-eb12-4d7c-b80e-915d867efa39 |
| 2.64 | no | https://logs.px4.io/plot_app?log=7bfbe6d4-4cc8-4cbd-8d69-e0669a764e10 |

**Specifications**

Ubuntu 16.04 LTS
ROS Kinetic
Gazebo 7.0.0
commit 891b2f6

LorenzMeier, 2017-08-20:
Thanks! Any addition insights into this?

nkhoit, 2017-08-24:
Playing with it, but it's pretty easy to reproduce.
Here's a video of it in action: https://www.youtube.com/watch?v=7MbafH5Xy1c

anassinator, 2017-08-24:
The only other insights I have at the moment are that this issue is not unique to this location and can be reproduced in all initial home positions I’ve tried. This makes me think it’s unrelated to #6843, but I tried the fix in PX4/sitl_gazebo#96 in case and it doesn’t seem to affect it. Not sure what to make of this yet though.

anassinator, 2018-03-27:
This is actually still an issue on latest master.

closed 2019-02-12 by stale[bot]