PX4/PX4-Autopilot #7794: Missing Mixer for Fixed-Wing HIL Simulation

acfloria, 2017-08-15:
I tried to run the HIL simulation using the XPlane10 simulator. The airplane model is the HILStar downloaded from the Pixhawk webpage. The Pixhawk is flashed with the latest release (v1.6.5) and using the HILStar(XPlane) airframe. The stable release v3.2 of QGC is used. 

When running the simulation according to this [Guideline](https://pixhawk.org/users/hil) the data from the XPlane simulator is received on the Pixhawk. However, it is not possible to fly the airplane with a remote control. The reason is that the AERT mixer is missing, it was deleted in [commit](https://github.com/PX4/Firmware/commit/8ff40d80485ecb4cfb772dd8f543d67bbfd4e99b?diff=unified)  previous to the latest firmware release. By just adding the [latest version](https://github.com/PX4/Firmware/blob/b42c1123a2c0e8cf5e0481aff9bdd0e7d038d546/ROMFS/px4fmu_common/mixers/AERT.main.mix) of the mixer to the firmware the HIL simulation works as expected.

Was the AERT mixer deleted accidentally or is there a way to run the HIL simulation without adding the mixer manually to the firmware?

dagar, 2017-08-15:
I've been meaning to fix this. https://github.com/PX4/Firmware/pull/7620

I'll find time later today to go through it with the latest PX4 + QGC and make sure everything is working again.

acfloria, 2018-01-31:
It is still missing, should I just add the mixer again?

acfloria, 2018-02-14:
Fixed by [#7794](https://github.com/PX4/Firmware/issues/7794)

closed 2018-02-14 by acfloria