PX4/PX4-Autopilot #7795: Strange oscillations in high speed quadcopter

GregoireH, 2017-08-15:
**Bug Report**

Hello,
I have struggling with this for a while and decided to share it, as I don't see any reasonable reason for this issue yet.

Here is the log
https://logs.px4.io/plot_app?log=83cacffd-e2e1-44ce-9e0b-13ced6287726

I have been **flying successfully at 4m/s**, but as soon as I speed up the drone I get some oscillations where the drone almost stop in the air and then start flying properly again.
It looks like something is wrong and it breaks to stabilize or something else and then start flying again.

The log is at 8m/s and 10m/s cruise speed.
Firmware release v1.6.3

I have really been trying different things unsuccessfully to remove those weird motions.
I am sorry if I missed something obvious.

Thanks in advance for your comments and help on that.
I remain at your disposal in case you need further information.
Best

GregoireH, 2017-08-15:
Maybe I should add that I can fly properly controlling the quad-copter remotely in position mode and stabilize mode at high speed without any issue. Those issue come in waypoint navigation only.

Best

lakshaydang, 2017-08-18:
You are able to solve this issue ? Is it a firmware bug ?

GregoireH, 2017-08-21:
Nop, not yet unfortunatelly...

closed 2018-02-21 by PX4BuildBot