PX4/PX4-Autopilot #7800: pixhawk 2.1 firmware px4fmu-v3_default  has motor_test command problem

gjkunde, 2017-08-16:
Thank you @LorenzMeier for adding motor_test in response to #7760,
alas - while the command is now part of nsh for the lastest px4fmu-v3_default
it does not show the desired result. If a scope is attached to the signal output
of the pixhawk 2.1 all eight channels show 350 Hz pulses with about 900 us with,
not matter that the motor_test test -m [0..7] - p [1..100] sets and corretly displays as
set result ... do I have to disable any other running process ? When working
with the intelfc code the motor_test command had the desire result of selecting
the power lever between 1 and 100 percent without any further ado.

LorenzMeier, 2017-08-16:
You need to (as in any Pixhawk system) disable the safety before you arm. So in this case you need to press the safety switch on the GPS puck before running the command.

bkueng, 2017-08-16:
`motor_test` is not supported by all boards (or rather output drivers). On the Aero it is (it's using `tap_esc`), on Pixhawk it's not. See https://dev.px4.io/en/middleware/modules_command.html#motortest.
You can use the `pwm` command instead: https://dev.px4.io/en/middleware/modules_command.html#pwm

closed 2017-08-20 by LorenzMeier