PX4/PX4-Autopilot #7801: vmount still causes board freeze-ups

ndepal, 2017-08-16:
This time I found out the hard way: Flew a mission with `MNT_MODE_IN 2`, `MNT_MODE_OUT 1`. Quad took off, kept rising way above setpoint, to about 15m instead of 5m. Then hovered there. No control possible via RC or QGC. Had to wait for battery to run out and the quad to come crashing down.

This happened with a slightly modified version of 55a2930cdb1e3162751d78193400a0983131238c.

The good news is that it appears to be reproducible in HIL.

## Steps to reproduce
- Configure a PixRacer as HIL quad
- set `MNT_MODE_IN 2`, `MNT_MODE_OUT 1` (other combinations don't appear to have this problem)
- start the HIL simulation
- Apparently coinciding with the setting of the home position, the board freezes up.

Increasing the stack size from 1900 to 4000 made no difference, so I doubt it's just a stack issue.

@bkueng , do you have any more good ideas?

ndepal, 2017-08-16:
It looks as if the main `vmount` loop runs much quicker once home is set, as evidenced by placing print statements in the main loop.

Rate-limiting the loop [here](https://github.com/PX4/Firmware/blob/55a2930cdb1e3162751d78193400a0983131238c/src/drivers/vmount/vmount.cpp#L322) the same way it is done [here](https://github.com/PX4/Firmware/blob/55a2930cdb1e3162751d78193400a0983131238c/src/drivers/vmount/vmount.cpp#L347) seems to help.

bkueng, 2017-08-16:
Sorry for the crash and that you have to go through this.

The cause was another type of bug, resulting in a busy loop as well: https://github.com/PX4/Firmware/pull/7805

dagar, 2017-08-17:
Consider changing the priority of vmount? https://github.com/PX4/Firmware/blob/master/src/drivers/vmount/vmount.cpp#L446

It's currently higher than navigator and equal to commander.

https://github.com/PX4/Firmware/blob/master/src/platforms/px4_tasks.h

closed 2017-08-17 by LorenzMeier

ndepal, 2017-08-17:
Tested in HITL. Did not get a system freeze.
I agree that the scheduling priority should be reduced. What's reasonable? `SCHED_PRIORITY_DEFAULT`? Is there any flight critical (control/estimation/rc link) component that could still be affected if there were still a busy loop in `vmount`?

dagar, 2017-08-17:
SCHED_PRIORITY_DEFAULT sounds fine to me in this case, unless you want to specifically call it out in px4_tasks.h as something new.