PX4/PX4-Autopilot #7815: NuttX enable CONFIG_MEMSET_OPTSPEED

dagar, 2017-08-18:
Setting CONFIG_MEMSET_OPTSPEED (Optimize memset() for speed) on NuttX decreases cpu usage by 2-3% on a pixracer configured as a multicopter. Most of the decrease is in the attitude controller and EKF (probably some of our excessive/unnecessary struct initialization). There's a small increase in flash.

Investigate CONFIG_ARCH_MEMSET as well.

dagar, 2017-08-22:
TODO: write basic testing to verify nuttx memset options.

davids5, 2017-08-24:
@dagar Did you want this in before the NuttX upgrade?

dagar, 2017-08-24:
To coincide with heavier NuttX testing?

davids5, 2017-08-24:
Yes it is 2 birds with one stone.  But the rigorous testing my not be needed:  It turns out there is no arch specific equivalent for memset like memcpy CONFIG_ARMV7M_MEMCPY - so my fears of a similar error in the calling convention are not ill founded.

I have lit CONFIG_MEMSET_OPTSPEED and CONFIG_MEMSET_64BIT on master_nxphlite do you want it on master ?

dagar, 2017-08-24:
I think it can wait, it'll be a nice little bonus with the update. Out of curiosity is there a way to bulk change all nuttx-configs for a single change like this or do you have to do each manually?

Fixed by https://github.com/PX4/Firmware/commit/ff8246fdee4f115e9e8f3606a1e603ed9fdef275

closed 2017-08-24 by dagar

davids5, 2017-08-24:
@dagar For thing like this, I do a scoped search and replace and have a regex tool that converts the modified files list to a list of commit commands. For refreshes it is more a manuals process, running make oldconfig on each target.