PX4/PX4-Autopilot #7817: Rename VDev to CDev

dagar, 2017-08-18:
This pattern is becoming quite common. Would it make sense to rename VDev to CDev or otherwise find an acceptable common name?

[image]

Do we have any basic documentation that specifies when you need to use px4_{open, close, read, write, poll} vs the regular version? There's lots of confusion here.

@davids5 @mcharleb Thoughts?

mcharleb, 2017-08-18:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBDevices.hpp#L71

You can't have both VDev and CDev be called the same thing when they are used concurrently in NuttX

mcharleb, 2017-08-18:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBDevices.hpp#L49

mcharleb, 2017-08-18:
CDev is an in-kernel character driver. VDev is a userspace virtual device.

dagar, 2017-08-18:
Yes, but why does that matter from the perspective of PX4? VDev isn't even built on NuttX.

The other thing to keep in mind is that for the vast majority of devices the /dev entry is barely used within PX4 and never by the rest of the system. Nearly all drivers directly publish output as uORB messages, and for the next driver framework iteration I'd like to drop the character device entirely.

mcharleb, 2017-08-18:
VDev IS used on Nuttx via the links I provided above.

mcharleb, 2017-08-18:
Based on your proposed roadmap, I'd leave VDev, and try to remove all uses of CDev

dagar, 2017-08-24:
That's still NuttX using CDev in place of VDev. They already provide nearly an identical interface, and merging them would allow us to fairly easily use the existing NuttX drivers cross platform.

Here's why I find this compelling. https://github.com/PX4/Firmware/pull/7864

dagar, 2017-09-06:
Done.

closed 2017-09-06 by dagar

mcharleb, 2017-09-06:
@dagar The device lookup code should be profiled since it's an array lookup. Shold be ok as long as number of devices is small.

dagar, 2017-09-06:
@bkueng recently changed it to a map for devices. We could do the same for the file map (still a static array).

bkueng, 2017-09-06:
Most operations are using `get_vdev`, which has constant runtime, so that's good. `px4_open` is linear, but it's only heavily used during startup, for a couple of hundred items, so I think that's ok for now as well.