PX4/PX4-Autopilot #7818: Support Artistic Style Version 2.05.1 (Ubuntu 16.04)

mcharleb, 2017-08-18:
When building on Ubuntu 16.04 the format checks fail because astyle 2.05.1 wants to fix the following:

```
# Space after {
-	int _airspeedSub{-1};
-	int _vehicle_status_sub{-1};
+	int _airspeedSub{ -1};
+	int _vehicle_status_sub{ -1};

# Space after function before (
-		delete (_log_writer_file);
+		delete(_log_writer_file);

# Indenting comments
-/**
- * Multi-rotor mixer for pre-defined vehicle geometries.
- *
- * Collects four inputs (roll, pitch, yaw, thrust) and mixes them to
- * a set of outputs based on the configured geometry.
- */
-class __EXPORT MultirotorMixer : public Mixer
+	/**
+	 * Multi-rotor mixer for pre-defined vehicle geometries.
+	 *
+	 * Collects four inputs (roll, pitch, yaw, thrust) and mixes them to
+	 * a set of outputs based on the configured geometry.
+	 */
+	class __EXPORT MultirotorMixer : public Mixer

# Adding braces
-		case 'd': if (nullptr != myoptarg) strcpy(_options.device, myoptarg); break;
+		case 'd': if (nullptr != myoptarg) { strcpy(_options.device, myoptarg); } break;

```
Builds have to be done in a Docker container for the format checks to pass.

mcharleb, 2017-08-18:
It seems that astyle 2.05.1 is broken and was fixed in later versions (I tested 3.0.1). I would remove 2.05.1 from the approved list in  Tools/check_code_style_all.sh

LorenzMeier, 2017-08-18:
Happy to merge a PR - is 3 available in 16.04?

mcharleb, 2017-08-18:
No, its not. I just downloaded and built the code.

mcharleb, 2017-08-18:
https://github.com/PX4/Firmware/pull/7819

closed 2017-08-19 by LorenzMeier