PX4/PX4-Autopilot #7820: VTOL Tilt Rotor arm/disarm and tilt control bug?

sparky002, 2017-08-18:
Using the latest px4 build as of a few days ago.

Is there any reason I would lose the ability to disarm my vtol tiltrotor aircraft from the rc transmitter after performing a transition to fixed wing mode as a test on the ground in an armed state? In multi-copter mode I can disarm though. I can also disarm during the transition phase in either state.

dagar, 2017-08-19:
We could verify in a log, but I'm guessing this is due to the recent land detector changes which now only report not landed while in FW mode.
@sanderux  FYI

sparky002, 2017-08-20:
I ran a test in fixed wing mode where I powered on the aircraft, pressed the safety button, unkilled the aircraft via a switch on the transmitter, armed the aircraft via a switch on the transmitter, actuated some aileron controls, spun up the motors, spun down the motors, then tried disarming via the same switch i armed with on the transmitter.
[14_08_15.ulg.tar.gz](https://github.com/PX4/Firmware/files/1237192/14_08_15.ulg.tar.gz)

dagar, 2017-10-08:
I may have misunderstood. Disarming via RC is intentionally disabled for FW so it doesn't happen in flight accidentally.

closed 2018-02-21 by PX4BuildBot