PX4/PX4-Autopilot #7833: vtol: thrust_min

Stifael, 2017-08-21:
Currently the Multi-copter land-detector relies on thrust values which represent the actual system thrust limits with some safety margin. The hover thrust and minimum thrust are used to determine if the vehicle is in a low thrust state (indication for demanding downward thrust: [here](https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L394) ).
In order to make the detection work as expected, it is desired that the hover thrust [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L72) is set as close as possible to the actual hover thrust, which is not necessarily 0.5. The minimum thrust (https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L53) should also be set low. The larger the difference between low and hover, the better the land-detection will work. 

I have learned that some vtols use large minimum thrust to keep the vehicle up during transition (please see comment [here](https://github.com/PX4/Firmware/issues/7768#issuecomment-321752131). The land detector is not really designed for that and there is no guarantee that a false-positive land-detection occurs if the minim thrust is large. The land-detection should be designed rather conservative to prevent any false-positive, otherwise we need to find a different solution for vtol.

To find a better solution than the current vtol state, I would like to discuss following points:

- MPC_THR_MIN: is this comment [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L42) actually correct? is it a "just" auto minimum thrust (I actually expected that it is a global, minim thrust for any mode)? Is it considered as a tuning parameter? I thought that this parameter is a lower limitwith safety margin (as opposed to a tuning parameter)
- MPC_MANTHR_MIN: this parameter is only used in the land-detector. What exactly is the purpose of this parameter? Why is it defined in the position-control-parameter file instead of the land-detector parameter file?
- Is it really the best solution to set the global parameter MPC_THR_MIN to 0.35 for vtols? If the thrust need to stay high during transition, could that not be solved locally? In my experience with controls, a good controller would consider the limits in the optimization. However, since that is computational more expensive and also much harder to do, controllers are written as if they have no limits and then afterward constrain the output. However, from a control perspective it is usually not the best idea to modify the limits to achieve a value that is anyway within the allowed range. Any value within max and min should be reachable by the control output without adjusting the limits. 

related issues and pr:
https://github.com/PX4/Firmware/issues/7768#issuecomment-321752131
https://github.com/PX4/Firmware/pull/7772#issuecomment-321913832

LorenzMeier, 2017-08-21:
@Stifael Let's discuss in person later this week. Manual thrust is used in the position controller: https://github.com/PX4/Firmware/blob/b620da8f01e66ec7653e367f86597629cac160e9/src/modules/mc_pos_control/mc_pos_control_main.cpp#L400

sanderux, 2017-08-21:
>  have learned that some vtols use large minimum thrust to keep the vehicle up during transition
Actually this is set to prevent loss of attitude when landing. Using pusher assist can cause a vtol to get lift from its wing when landing in a nose wind. 

To counter this i have disabled pusher assist during autonomous landings but that does not fully mitigate the problem. 

A proper VTOL land detector should likely be considering the following;
- If in auto mode, only activate on a LAND setpoint or when rc controlled only when stick is down
- increase the minimum time that no downward velocity may be felt to about 4 seconds
- attempt to go below min thrust if vehicle is spinning at minimum thrust for some time and very little has changed to the altitude and attitude. if it follows min thrust it is likely landed

Stifael, 2017-08-21:
@sanderux, I think that is straight forward to do in the vtol class. 
But nonetheless, I still would like to discuss the large minimum thrust value of 0.35, and even more if it is an attitude problem that you try to solve when demanding downward movement.

closed 2018-02-21 by PX4BuildBot