PX4/PX4-Autopilot #7843: PX4 1.6.5 Does not hold heading on Position mode.

tubeme, 2017-08-22:
It is Plain Airplane Frame 2100

On Position mode the plane does not hold its heading and the wind rotates it tail so we have to make corrections all the time.

On Return mode it flyies in a perfect line. 

In the Log I see there is no Yaw SP during Position mode.

While in Return mode it has Yaw SP.

https://logs.px4.io/plot_app?log=dbeb21a4-dcf5-4e2a-8e05-ca591b1c1da5

Any ideas?

dagar, 2017-08-22:
Hi Vasil,

FW position control mode only locks onto a heading when your roll and yaw sticks are centered (in the deadzone).

Here you can see the roll setpoint changing with your roll stick movement.

[image]

AlexisTM, 2017-08-22:
I believe there could be an offset on the remote. 

Set a zero offset on the remote and click on copy trims, to be sure it does not come from a soft trim of the Pixhawk.

dagar, 2017-09-08:
Not a firmware issue? Please reopen if I'm incorrect.

closed 2017-09-08 by dagar