PX4/PX4-Autopilot #7846: Assisted and Mission Mode with sudden roll and pitch actuation blowing past limits

sparky002, 2017-08-22:
https://review.px4.io/plot_app?log=a6c22f8b-072f-4bea-99f0-021ed25ea2ca
-Small electric Piper Cub aircraft
-Latest px4 build
-Airframe: | Multiplex EasystarStandard Plane (2100)
-Hardware: | PX4FMU_V2
-External gps/mag
-Airspeed sensor free of disturbances

Did entire flight in Rattitude mode but similar situation described below happened in a previous rattitude mode flight and in Mission mode too. Did not happen in stabilized mode.

Took off from a grassy runway to 3ish meters max, did a coordinated turn to the right, turned hard, levelled out, turned to the right again, levelled out and suddenly the aircraft rolled to the right and pitched down bypassing the limits set to 30 degrees and 30 degrees/sec rate for manual limits and controller limits in QGroundControl. Pixhawk seemed to obey them in stabilize mode and throughout other portions of the rattitude and mission flights. Was not going slow enough for a stall and it seemed the Pixhawk was commanding these controls.

The pixhawk and sensors were well secured and did proper calibrations beforehand. Double/triple checked all the wiring, connections and hardware.

Any ideas?

LorenzMeier, 2017-08-22:
The vibration levels measured were certainly out of limits:
https://review.px4.io/plot_app?log=a6c22f8b-072f-4bea-99f0-021ed25ea2ca#Nav-Raw-Acceleration

sparky002, 2017-08-22:
The Pixhawk was rigidly mounted to the foam body with no extra damping. That would explain the vibration levels being to high. Do you know what the vibration limits for the Pixhawk are or what the typical range we should be operating in for the Pixhawk? Useful information to be aware of when diagnosing future flights, in lab testing and damping improvements. 
-Thank you for pointing that issue out.

sparky002, 2017-08-24:
Recreated the flight in Xplane HITL. Rattitude mode caused the aircraft to violently roll over when the transmitter stick reached a certain position. All the other flight modes acted as expected.

dagar, 2017-12-06:
@sparky002 are you sure that wasn't rattitude mode working as intended (acro mode past a certain stick position)? Please recreate and post a flight log if you think it's still an issue.

closed 2017-12-06 by dagar