PX4/PX4-Autopilot #7852: Nuttx Upgrade for 1.7 Release

davids5, 2017-08-23:
I will request @PX4TestFlights to begin testing when the branch is ready.

1. master_nxphlite branch needs complete flight testing on all HW
2. AI:Need to insure that PX4Test has all HW targets
3. AI:Update Test Doc with coverage matrix for HW
4. AI: @LorenzMeier - send email documenting the "Support levels" and request HW from vendors for PX4Test
5. Document the support level for each board/config in readme.

r0gelion, 2017-08-23:
@davids5 

Based on the [Supported Hardware List](https://github.com/PX4/Firmware#supported-hardware) this is what we have and don't have:

- Snapdragon Flight
  - 1 x QAV250 - PWM ESCs
  - 1 x 200qx - UART ESCs (Qualcomm 4-in-1)
- Intel Aero Ready To Fly Drone
  - 2 drones
- Raspberry PI with Navio 2
  - **None**
- Parrot Bebop 2
  - **None**
- FMUv2.x
  - Pixhawk
    - 1 x DJI F450 - Pixhawk without silicon errata
  - Pixhawk Mini
    - 1 x DJI F450
    - 1 x Phantom Flying Wing
  - Pixfalcon
    - **None**
- FMUv3.x Pixhawk 2
  - **None**
- FMUv4.x
  - Pixracer
    - 1 x DJI F450
  - Pixhawk 3 Pro
    - 1 x DJI F450
    - 1 x VTOL Falcon Vertigo (temporary while we repair or replace Dropix autopilot)
- FMUv5.x (ARM Cortex M7, future Pixhawk)
  - **None**
- STM32F4Discovery
  - **None**
- Gumstix AeroCore (v1 and v2)
  - **None**
- Airmind MindPX V2.8
  - **None**
- Airmind MindRacer V1.2
  - **None**
- Bitcraze Crazyflie 2.0
  - **None**

TSC21, 2017-08-24:
I think we are missing the Aerotenna OcPoC there

davids5, 2017-08-24:
@r0gelion 

There are 2 goals here:

1. To get the test team more hardware for testing. 
Please grab the uinque hw types listed  [here](https://github.com/PX4/Firmware/tree/master_nxphlite/cmake/configs)  and update the list above.

2. We will then qualify each HW with (a yet to be determined) "Level of Support"  @LorenzMeier will be defining the levels of support.

r0gelion, 2017-08-24:
@davids5 

The list is updated.

closed 2017-10-05 by LorenzMeier