PX4/PX4-Autopilot #7853: AIRSPEED SENSOR MISSING in rover mode

gpldecha, 2017-08-23:
When using the rover, the following message comes up showing that the airspeed sensor is missing.
This does not seem necessary when using a rover airframe.

APM: PREFLIGHT FAIL: AIRSPEED SENSOR MISSING

TSC21, 2017-08-24:
@gpldecha as a provisional fix, try setting `CBRK_AIRSPD_CHK` to `162128`.

TSC21, 2017-08-24:
The problem is that `GroundRoverPositionControl` relays on the `ECL_L1_Pos_Controller`, which verifies for the presence of airspeed data. It is on the TODO list reformulate this same matter, as the UGV controllers came basically from the FW controllers.

dagar, 2017-10-08:
An acceptable fix for now is having CBRK_AIRSPD_CHK default to 162128 for all UGV.
Slightly longer term we need commander to be aware of ugv and not just fw/mc/vtol.

closed 2018-02-21 by PX4BuildBot

dlarue, 2018-12-15:
bump, issue still there.