PX4/PX4-Autopilot #7858: make posix_sitl_default gazebo failed on Ubuntu 16.04

whyscience, 2017-08-24:
make posix_sitl_default gazebo failed on Ubuntu 16.04, it is OK on Ubuntu 14.04, very strange.

> [ 24%] Built target lib__rc__rc_tests
Scanning dependencies of target gazebo_opticalFlow_plugin
[ 80%] Building CXX object CMakeFiles/gazebo_opticalFlow_plugin.dir/src/gazebo_opticalFlow_plugin.cpp.o
/home/eric/all_code/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp: In member function ‘void gazebo::GeotaggedImagesPlugin::handle_message(mavlink_message_t*)’:
/home/eric/all_code/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:361:50: error: too many arguments to function ‘uint16_t mavlink_msg_command_ack_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint16_t, uint8_t, uint8_t)’
                                        msg.compid);
                                                  ^
In file included from /opt/ros/kinetic/include/mavlink/v2.0/common/common.h:1165:0,
                 from /opt/ros/kinetic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/eric/all_code/Firmware/Tools/sitl_gazebo/include/gazebo_geotagged_images_plugin.h:32,
                 from /home/eric/all_code/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:18:
/opt/ros/kinetic/include/mavlink/v2.0/common/./mavlink_msg_command_ack.h:89:24: note: declared here
 static inline uint16_t mavlink_msg_command_ack_pack_chan(uint8_t system_id, uint8_t component_id, uint8_t chan,

The reason is d640d1aaf15e5bba3e579642ee277fa52690d041 
Root cause is https://github.com/PX4/sitl_gazebo/commit/ef7d0aa81321521a8b2496131d636858b95b6068

TSC21, 2017-08-24:
This issue was identified yesterday. Are you using `sitl_gazebo` out of the Firmware directories?

whyscience, 2017-08-24:
@TSC21  No, like this
`~/src/Firmware$ make posix_sitl_default gazebo`

TSC21, 2017-08-24:
The problem is that it looks for the ROS Mavlink release, which currently does not track the latest mavlink release.

whyscience, 2017-08-24:
Yes, I just install ROS Mavlink and source `source ~/catkin_ws/devel/setup.bash`

TSC21, 2017-08-24:
> @TSC21  No, like this
`~/src/Firmware$ make posix_sitl_default gazebo` 

Then it should be addressed by, https://github.com/PX4/sitl_gazebo/blob/ef7d0aa81321521a8b2496131d636858b95b6068/CMakeLists.txt#L131. This you update the mavlink submodule? Are you using current master?

TSC21, 2017-08-24:
> Yes, I just install ROS Mavlink and source `source ~/catkin_ws/devel/setup.bash`

As I said, the problem is not you not having the latest ROS Mavlink. The problem is that the latest ROS Mavlink is not synced with latest Mavlink upstream. It needs to be released first on the ROS packages.

whyscience, 2017-08-24:
I am using current master
Yeah, I got it.

TSC21, 2017-08-24:
Nevermind. I can also reproduce this issue, even using latest master. CMakeList should be updated so to track the latest Mavlink available on the system instead of the ROS version. I will try to find a solution to this. Thanks for confirming the issue @whyscience

whyscience, 2017-08-24:
My pleasure

TSC21, 2017-08-24:
You can always downgrade `sitl_gazebo` for a previous commit so to use it right now. This will be fixed in the mean time

whyscience, 2017-08-24:
OK, I already did that

Maunsi, 2017-08-24:
I'm having the same problem and I'm fairly new to px4-ros-everything, how do i a downgrade to a previous commit to be able to use it?

TSC21, 2017-08-24:
Use `git reset --hard 58fb126` on `sitl_gazebo` (`<Firmware_dir>/Tools/sitl_gazebo`)

ChristophTobler, 2017-09-18:
There's another error now:
```
error: ‘DODECA_TOP_COX’ is not a member of ‘MultirotorGeometry’
   geometry = MultirotorGeometry::DODECA_TOP_COX;
```

bkueng, 2017-09-18:
> There's another error now:

You can try with: `rm <build_dir>/src/modules/systemlib/mixer/mixer_multirotor.generated.h`

ChristophTobler, 2017-09-18:
>You can try with: rm <build_dir>/src/modules/systemlib/mixer/mixer_multirotor.generated.h

@bkueng Still the same error...

See also https://github.com/PX4/Firmware/issues/7978

ChristophTobler, 2017-09-19:
The file mixer_multirotor.generated.h still existed outside of the build folder and therefore the file was not generated again. Deleting it in src/modules/systemlib/mixer/mixer_multirotor.generated.h fixed the issue. @bkueng Thanks!

closed 2017-12-16 by whyscience