PX4/PX4-Autopilot #7860: Manual Overwrite not allowed in Stabilized Flight Mode

acfloria, 2017-08-24:
While testing the v1.6.5 of the PX4 Firmware I discovered that in the stabilized mode the manual overwrite is not allowed:
[https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3895-L3896](url)

The result is that if the FMU crashes the mixer uses enters the MIX_FAILSAFE mode and outputs zero commands. Therefore the plane will crash. By allowing the OVERRIDE the pilot could at least try to land the plane in the manual mode.

dagar, 2017-08-24:
(Duplicated on discuss: http://discuss.px4.io/t/failsafe-in-case-of-an-fmu-crash-for-an-airplane/4209/7) It seems like you'd want the servos to go to the failsafe values unless you were already in manual (then continue in manual mode). You should also be able to switch into manual mode in that state. Is the px4io capable of interpreting mode switches? I need to review the code.

TSC21, 2017-08-24:
May this become handy @dagar? https://github.com/PX4/Firmware/pull/4177

dagar, 2017-08-24:
Looks like the px4io is only aware of the "old" style mode switch configuration. https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp#L1615

We should really kill that off and have only one configuration.

@TSC21 no I don't believe so, but it's interesting.

TSC21, 2017-08-24:
> We should really kill that off and have only one configuration.

Yeah checking the same also. We should level the RC channel config for all drivers/modules.

acfloria, 2017-08-25:
During testing I checked that the value of RC_MAP_MODE_SW is set correctly. 
When simulating a crash of the FMU by stopping the attitude controller this enabled me to switch to the manual mode and still control the plane. However, in a total crash of the FMU (either by coding an error or holding the restart button) I was not able anymore to change to the manual mode.

acfloria, 2017-09-05:
I looked a bit into the code and I wasn't able to find the functionality for mode switching in the px4io firmware. So I have two questions:

1. Is there a functionality implemented to switch the modes in the px4io based on the switch RC_MAP_MODE_SW

2. Why is the manual override not ok in the stabilized mode?
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3909

dagar, 2017-09-05:
It's handled on the px4io driver side that runs on the fmu. https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp#L1615

Manual override is not okay for multicopters in stabilized mode. I agree it's better than nothing for planes.

acfloria, 2017-09-05:
The manual override is by default disabled for vtol and rotary wings, so setting it explicitly to false should only affect fixed wing planes:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3877

proficnc, 2017-09-05:
Even on a quadplane, a manual override attempt is better than a crash.... more complex... but better...

acfloria, 2018-01-31:
@dagar couldn't we allow the manual override in the stabilized mode until we have proper solution for the mode switch in the px4io firmware?

acfloria, 2018-02-09:
Fixed by [#8062](https://github.com/PX4/Firmware/pull/8062)

closed 2018-02-09 by acfloria