PX4/PX4-Autopilot #7861: Mixer of Bormatec Maja not working in Manual Override

acfloria, 2017-08-24:
While testing the v1.6.5 of the PX4 I found the following issue:

During normal flight operation everything works as expected. In case of an FMU crash in the manual flight mode the OVERRIDE flag is correctly set in the px4io. However, in this mode the outputs for MAIN 1, MAIN 3 through MAIN 5 are correctly set. But the MAIN 2 (right aileron) output remains always at 1500 which corresponds to a zero deflection.

This behavior can be solved by removing the lines for the airbrakes ([L24](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/mixers/AAERTWF.main.mix#L24), [L29](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/mixers/AAERTWF.main.mix#L29)), or changing the index from 6 to a value lower than 4. However, I was not able to identify why an index value of 6 or higher breaks the mixing.

See also this topic in the [forum](http://discuss.px4.io/t/failsafe-in-case-of-an-fmu-crash-for-an-airplane/4209/5).

dagar, 2017-08-24:
I'm not sure what the root cause is, but we can safely remove the airbrakes entries from the mixer.

LorenzMeier, 2017-08-26:
Thanks for the fix!

closed 2017-08-26 by LorenzMeier

acfloria, 2017-09-05:
@LorenzMeier 
This was more of a temporary fix to the issue as somebody might want to use the flaperons or any other input with an control group index > 5 in the future.

I do not really know the px4io code but I think the issue is caused by the px4io during loading or applying the mixer. Could maybe someone with more knowledge of px4io code trace down the cause of this issue?