PX4/PX4-Autopilot #7863: OFFBOARD loss with logging freeze

Shpiler, 2017-08-24:
Yesterday i met some strange behavior on my drone. We had a mission to takeoff, fly and land in another location. Everything in OFFBOARD mode using RPi and MAVROS. 

After takeoff the drone suddenly lost offboard mode for a second. It could be our code glitch... BUT! At the same moment ALL flight data hasn't been logged! Too strange to be a coincidence. Look at the logs. Also here is some our logs with flight mode got from mavros topic and timestamp (microseconds). You can see POSCTL in our log but not in PX4 log. Also COM_OF_LOSS_T was set to 4 seconds and that makes things even more strange.  

[image]
[image]
[image]
[image]

[13_15_19.ulg.zip](https://github.com/PX4/Firmware/files/1249566/13_15_19.ulg.zip)

TSC21, 2017-08-24:
can you please post the log on logs.px4.io?

Shpiler, 2017-08-24:
https://logs.px4.io/plot_app?log=a410c5bb-478b-4146-8054-8f26745eacae

TSC21, 2017-08-25:
The only thing that comes to me right now is that it could be a temporary powering issue

Shpiler, 2017-08-25:
How could it be powering issue without influence to flight control or any other subsystem? At the same moment RPi received at least one mavlink package, and drone's orientation / motor's sound was normal and stable.

TSC21, 2017-08-28:
> How could it be powering issue without influence to flight control or any other subsystem?

Power issues or spikes may usually result on logging drops. I already saw that happening on other logs I've seen. Did you reproduce this on other experiments during testing? Or did you just saw this happening once?

Shpiler, 2017-08-29:
> Did you reproduce this on other experiments during testing?

Yes, it was logged at least once on the different hardware. Not sure I can find that file, but it was almost the same. Today we've lost offboard mode once again, unfortunately with no SD card inserted. Will continue testing. 
I don't care so much about a few seconds of missing log data, but offboard losses are critical since we are going to fly without RC at all or, at least, far away out of visual range.

TSC21, 2017-08-29:
Did you make sure your telem link is stable? Can you show me the output of the MAVROS terminal? Also the output of `rostopic echo -n1 /diagnostics`? This of course during a test where you actually get this problem. What's the baud rate of your link?

closed 2018-02-21 by PX4BuildBot