PX4/PX4-Autopilot #7869: Failed to flash board through terminal and QT

CastleUAVStudio, 2017-08-26:
I tried to flash Pixhawk version:4 ID:9
so I do
`make px4fmu-v2_ekf2 upload`
successfully erase, program, verify, rebooting. However no '[100%] Built target upload' sign, and no welcome sound etc.
[image]
I tried on terminal, and tried on QT, same happened.
[image]

However I can use QGC to download and flash the board.

Really need a help here.

LorenzMeier, 2017-08-26:
Please use `make px4fmu-v2_default upload` instead. The EKF2 target is deprecated and will be removed.

TSC21, 2017-08-26:
What is your autopilot hw? Pixhawk, Pixhawk 2, Pixhawk Mini, PixRacer,...?

closed 2017-08-26 by LorenzMeier