PX4/PX4-Autopilot #7870: Failed to flash board through terminal

CastleUAVStudio, 2017-08-27:
Thanks to @LorenzMeier for issue[#7869](https://github.com/PX4/Firmware/issues/7869),
 I tried`make px4fmu-v2_default upload` in terminal, still failed, same thing happened:
[image]
no '[100%] Built target upload' sign, and no welcome sound etc

Thanks for all the help.

CastleUAVStudio, 2017-08-27:
Tried couple times, QGC can download successfully, but terminal still can't work

dagar, 2017-08-30:
Have you tried a few different usb cables? Sometimes they're flaky.

CastleUAVStudio, 2017-08-30:
Thanks for your help! I will try this as soon as possible.

closed 2018-02-21 by PX4BuildBot