PX4/PX4-Autopilot #7871: VTOL MC attitude vs altitude

sanderux, 2017-08-27:
Some recent testing has exposed a problem specific to VTOL in MC altitude controlled modes.

https://logs.px4.io/plot_app?log=993613b8-0800-43ba-bb56-d8f2e98e377d
This was a flight in medium wind. In manual it tracks roll reasonably well, even when putting the large wing in a cross wind. In altitude it tracks very bad because it is also concerned with holding altitude. and saturates the motors.

VTOL's are prone to bad roll behavior in windy conditions as they generally carry a wing. When in altitude controlled modes i believe there should be a failsafe behavior that puts all available thrust towards attitude and return to control altitude as soon as attitude is back to acceptable. 

Currently we are addressing this by limiting the max angles in MC significantly but that's not ideal.

@dagar @bresch @AndreasAntener @LorenzMeier thoughts?

bresch, 2017-09-05:
@sanderux I just rewrote the attitude saturation handling part in the multirotor mixer. I'll make a PR so you can test it. It should solve those attitude instability problems.

sanderux, 2017-09-05:
Thats good, will be good to see how it copes

tubeme, 2017-09-07:
Very good idea. We noticed this as well. This concerns the Position mode as well.

RomanBapst, 2018-02-13:
@sanderux @bresch Is this solved?

bresch, 2018-02-13:
@RomanBapst Should be solved when https://github.com/PX4/Firmware/pull/7920 is working and merged.

sanderux, 2018-02-13:
I think this was already solved. closing for now

closed 2018-02-13 by sanderux