PX4/PX4-Autopilot #7872: Gazebo8 ignition protoc version conflict on Mac

julianoes, 2017-08-27:
The gazebo build on macOS Sierra with latest master fails because of a conflicting protoc version.

```
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/quaternion.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/quaternion.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/quaternion.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
In file included from SensorImu.pb.cc:5:
In file included from ./SensorImu.pb.h:33:
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/vector3d.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/vector3d.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/vector3d.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc
```

This is using the latest install instructions from [the Devguide](https://dev.px4.io/en/setup/dev_env_mac.html#common-tools):

```
brew tap PX4/px4
brew install px4-dev
# Optional, but recommended additional simulation tools:
brew install px4-sim
```

LorenzMeier, 2017-08-27:
Please report upstream: https://github.com/osrf/homebrew-simulation/issues/268

In the meanwhile install from source:

```brew reinstall --build-from-source gazebo8```

closed 2017-08-27 by LorenzMeier

julianoes, 2017-08-27:
@LorenzMeier how do I know that it is using osrf/homebrew-simulation?

LorenzMeier, 2017-08-27:
Looking at the recipe: https://github.com/PX4/homebrew-px4/blob/master/px4-sim.rb

julianoes, 2017-08-27:
Got it, thank you!