PX4/PX4-Autopilot #7880: pixhawk-pi connection

b-rohit, 2017-08-30:
I connected pixhawk and pi following the [tutorial](http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html). I don't find serial device node in `/dev ` directory and there is no change in `/dev` directory when I remove or connect.

jaypatrikar, 2017-08-30:
Did you disable the serial network use by os?

Regards,

Jay Patrikar
Integrated BT-MT
Dual Degree Student

Unmanned Aerial Vehicles Lab
Department of Aerospace Engineering
IIT Kanpur
Kanpur-208016

M: +91-7752846441

b-rohit, 2017-08-30:
yes I disabled it.

jaypatrikar, 2017-08-30:
I guess you are using PX4 and not Arducopter. In which case you have to enable the sys_companion parameter to a specific baud rate. U can also try using a FTDI/USB and see if the problem is from pixhawk side or rpi.  

Hope it helps!

b-rohit, 2017-08-31:
thanks @jaypatrikar . Yes I am using  PX4 and I have set `sys_companion` to 921600.

dagar, 2017-12-06:
Please reopen if this is still an issue.

closed 2017-12-06 by dagar