PX4/PX4-Autopilot #7891: UUID extension

simonegu, 2017-09-01:
During an analysis @dogmaphobic noticed that the MAVLink UUID sent was not complete this resulted in two units having the same UUID. Because the words passed to the MAVLink message.
https://github.com/PX4/Firmware/blob/c3b1ec8b246b123853943c68ce72a9876b399386/src/modules/mavlink/mavlink_main.cpp#L1339
I looked into how the UUID on PX4 for ST32 was constructed and noticed that the ordering of the different words was not as I would have expected.
https://github.com/PX4/Firmware/blob/bd60f87283f7938d89be7640afd9f02abcc72c03/src/platforms/px4_micro_hal.h#L95-L104
Only with the variable `PX4_CPU_UUID_CORRECT_CORRELATION` defined it would have been correct. Like this: 
https://github.com/PX4/Firmware/blob/bd60f87283f7938d89be7640afd9f02abcc72c03/src/platforms/px4_micro_hal.h#L79
with:
https://github.com/PX4/Firmware/blob/bd60f87283f7938d89be7640afd9f02abcc72c03/src/platforms/px4_micro_hal.h#L71-L73

I talked with @davids5 about this and other chip manufacturer even use 128bit UIDs and the solution would be to extend the UUID to 128bit to make it compatible with other manufacturer. For ST32 target it would look like this: `0000 IJKL EFGH ABCD`.

I also noticed that the `MFGUID` has the correct formatting. 

What I would propose is:
- [ ] Adapt the UUID with the correct and extended formatting.
- [ ] Add an UUID extended field in the MAVLink message to have a total length of 128bit.
- [ ] Deprecate the `MFGUID`.

Thoughts?

davids5, 2017-09-01:
The uuid32 API is the old (legacy) [byte ordering](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/common/board_common.h#L303) should be considered deprecated, [with this in mind](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/common/board_common.h#L310) - 

In the past I asked to remove the incorrect format but the notion was rejected.  The reason it was rejected may no longer be valid.  Please ask @LorenzMeier if it can come out in relase 1.7.0.

mfguid API is the new format in correct [byte order](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/common/board_common.h#L323) mfg refers to the Chip Manufacturer AKA arch: stm32, Kinetis, samv7

There were 2 problems the legacy : 
 1. Byte ordering (it was interpreted wrong and was thought to be only word addressable)
2. Length: Some new to PX4, chips have [128 bit](https://github.com/PX4/Firmware/blob/master_new_archs_nuttx_upgrade/src/platforms/px4_micro_hal.h#L160) uuids and the stm32 has a [96 bits](https://github.com/PX4/Firmware/blob/master_new_archs_nuttx_upgrade/src/platforms/px4_micro_hal.h#L82). uuid  - See [px4_micro_hal.h](https://github.com/PX4/Firmware/blob/master_new_archs_nuttx_upgrade/src/platforms/px4_micro_hal.h) PX4/Firmware on master_new_archs_nuttx_upgrade for the complete story. 

> I looked into how the UUID on PX4 for ST32 was constructed and noticed that the ordering of the different words was not as I would have expected.

`PX4_CPU_UUID_WORD32_UNIQUE_[L|M|H]` - the L M or H can be thought of is the speed that the digits increment as the chips are manufactured. 

The reason the incorrect usage was preserved here:
` msg.uid = (((uint64_t)uid[PX4_CPU_UUID_WORD32_UNIQUE_M]) << 32) | uid[PX4_CPU_UUID_WORD32_UNIQUE_H]; `
was also to not introduce a breaking change that would cause recalibration. 

To fix this just define `PX4_CPU_UUID_CORRECT_CORRELATION`. If we do NOT care about the history or do NOT want to support the option to have it wrong on forks, just remove the `ifdef PX4_CPU_UUID_CORRECT_CORRELATION` and keep the `then` correlation block. 

Moving forward we should define a set of PX4_ prefixed constants that define the API as 128 bits and use just the mfg API.

Then have the board code for all arch use a 128 bit field, right justify the 96 bit mfguuid in it.
[0000xxxxxxxxxxxx].  

Just keep in mind from arch to arch the value may not be globally unique.

davids5, 2017-09-01:
FYI @bkueng  Should we be proactive and goto 256 bits?

bkueng, 2017-09-04:
The adress space of 128 bits is already huge. IPv6 also uses 128 bit adresses and the phone IMEI has 16 decimal digits, so I'd say it's enough. But we should add 16 bits to distinguish the different archs.

simonegu, 2017-09-07:
@davids5 @bkueng should we extend the `MFGUID` to have the 128bit ? And take it as the default also in the logger?

davids5, 2017-09-08:
@simonegu see https://github.com/PX4/Firmware/pull/7935 

It adds PX4GUID with the 16 bit prefix for the arch to make it Globally unique from arch to arch.

ksschwabe, 2017-09-08:
One thing to keep in mind would be the UID that is reported by px_uploader.py when flashing a board.

davids5, 2017-09-08:
@ksschwabe - good point also the UID is used in UAVCAN and is 16 byte (FYI @pavel-kirienko)

closed 2018-02-21 by PX4BuildBot