PX4/PX4-Autopilot #7892: make posix gazebo_typhoon_h480 bug， ekf2: exceeded maximum number of file descriptors

whyscience, 2017-09-01:
For the latest master, here is the bug.
After some simple tests, it's mostly caused by `Add support for mavlink message DEBUG_VECT`
@TSC21 

> WARN  [drivers__device] ekf2: exceeded maximum number of file descriptors, accessing /obj/_obj_
ERROR [uorb] vehicle_global_position advertise failed
INFO  [platforms__posix__px4_layer] Backtrace: 9
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_backtrace+0x25) [0x57e805]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_open+0x145) [0x4227e5]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b47e]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b540]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4f802c]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4fa3de]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x57dad7]
INFO  [platforms__posix__px4_layer] /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba) [0x7fe6b53576ba]
INFO  [platforms__posix__px4_layer] /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7fe6b47fa3dd]
WARN  [drivers__device] ekf2: exceeded maximum number of file descriptors, accessing /obj/vehicle_global_position0
INFO  [platforms__posix__px4_layer] Backtrace: 10
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_backtrace+0x25) [0x57e805]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_open+0x145) [0x4227e5]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b32f]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b40a]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b540]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4f802c]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4fa3de]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x57dad7]
INFO  [platforms__posix__px4_layer] /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba) [0x7fe6b53576ba]
INFO  [platforms__posix__px4_layer] /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7fe6b47fa3dd]
WARN  [drivers__device] ekf2: exceeded maximum number of file descriptors, accessing /obj/_obj_
ERROR [uorb] vehicle_global_position advertise failed

TSC21, 2017-09-01:
What leads you to think it has anything to do with that?

whyscience, 2017-09-01:
I pull master, make posix gazebo_typhoon_h480, this warning occured
Roll back to just before `Add support for mavlink message DEBUG_VECT`, this error disappeared.

whyscience, 2017-09-01:
You may have a try

TSC21, 2017-09-01:
The only place I see that warning being called is on  src/drivers/device/vdev_posix.cpp, on which the debug msgs does not have any kind of interference.

whyscience, 2017-09-01:
Yeah, I know, but the warning just came out, I don't know either
Maybe I'm just wrong. Hope others give more test results.

dagar, 2017-09-01:
It's hard coded in vdev_posix.cpp to 300. 

SITL is obviously different than NuttX, but to give you an idea here's the FD usage on a pixhawk (last column).

	 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
	   0 Idle Task                21075283 42.899   604/  748   0 (  0)  READY  3
	   1 hpwork                     994965  2.207   816/ 1780 249 (249)  w:sem  7
	   2 lpwork                     193493  0.441   684/ 1780  50 ( 50)  READY  6
	   3 init                         8695  0.000  1272/ 2580 100 (100)  w:sem  3
	 151 log_writer_file            432361  0.956   560/ 1068  60 ( 60)  w:sem 43
	  12 dataman                      9648  0.000   712/ 1180  90 ( 90)  w:sem  4
	 112 gps                        100809  0.294  1040/ 1524 220 (220)  w:sem  5
	 114 sensors                   1726417  4.120  1464/ 1964 249 (249)  w:sem 14
	 116 fmu                        830886  1.913   816/ 1292 251 (251)  w:sem  9
	 125 commander                  585761  1.250  3016/ 3652 140 (140)  w:sig 33
	 126 commander_low_prio           1450  0.000   552/ 2996  50 ( 50)  w:sem 33
	 134 ekf2                      6829379 15.599  5104/ 5876 250 (250)  w:sem 15
	 136 fw_att_control            1269059  2.722  1024/ 1476 251 (251)  w:sem 12
	 138 fw_pos_ctrl_l1             348609  0.735   992/ 1676 250 (250)  w:sem 12
	 141 navigator                   94214  0.220   944/ 1772 105 (105)  w:sem 15
	 145 camera_feedback             70399  0.147   696/ 1572 115 (115)  w:sem  6
	 150 logger                    1991431  4.709  3000/ 3540 245 (245)  w:sem 43
	 203 top                            44  0.000  1296/ 1684 100 (100)  RUN    3
	 153 mavlink_if0               2333124  5.150  1736/ 2380 100 (100)  w:sig 40
	 154 mavlink_rcv_if0            203117  0.441  1808/ 2140 175 (175)  w:sem 40
	 165 mavlink_if1               1084538  2.354  1520/ 2380 100 (100)  w:sig 18
	 167 mavlink_rcv_if1            204881  0.441  1320/ 2140 175 (175)  w:sem 18
	 181 mavlink_if2               2591375  5.813  1648/ 2388 100 (100)  w:sig 43
	 184 mavlink_rcv_if2            950840  2.354  1320/ 2140 175 (175)  w:sig 43

If you have the time I'd dump the vdev list to make sure it seems reasonable, but I can believe we're exceeding 300 valid FDs.

whyscience, 2017-09-02:
Now it's like this ,after 962cdcf837d75ce2243ab635b916d55a07406754 

> WARN  [drivers__device] mavlink_if0: exceeded maximum number of file descriptors, accessing /obj/_obj_
INFO  [platforms__posix__px4_layer] Backtrace: 10
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_backtrace+0x25) [0x580145]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_open+0x145) [0x4228b5]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b54e]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b78e]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a14dd]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a151f]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a15a3]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a63a1]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4b0e5b]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x49d5fb]
WARN  [drivers__device] mavlink_if2: exceeded maximum number of file descriptors, accessing /obj/estimator_status0
INFO  [platforms__posix__px4_layer] Backtrace: 10
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_backtrace+0x25) [0x580145]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4(px4_open+0x145) [0x4228b5]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b3ff]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b4da]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x53b78e]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a14dd]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a151f]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a15a3]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4a63a1]
INFO  [platforms__posix__px4_layer] /home/eric/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4() [0x4b0e5b]
WARN  [drivers__device] mavlink_if2: exceeded maximum number of file descriptors, accessing /obj/_obj_

dagar, 2017-09-02:
Try increasing PX4_MAX_FD here -
 https://github.com/PX4/Firmware/blob/master/src/drivers/device/vdev_posix.cpp#L62

whyscience, 2017-09-02:
yeah try '#define PX4_MAX_FD 300/301/302/303', 303 is OK.
we should increase it to 305 at least

dagar, 2017-09-02:
@whyscience can you open a PR giving it a large boost? 350 is probably fine.

whyscience, 2017-09-03:
OK

dagar, 2017-09-03:
Fixed by https://github.com/PX4/Firmware/commit/2b714e079b97976da1b6f633699f7b79f1c01389

closed 2017-09-03 by dagar

huangwen0907, 2017-09-13:
@whyscience  You can have a try
> make posix_sitl_default gazebo_typhoon_h480

commit 2b714e07 (2017-09-03): driver: vdev_posix, increase PX4_MAX_FD (#7905)

 - avoid "exceeded maximum number of file descriptors" when "make posix gazebo_typhoon_h480"
 - closes #7892