PX4/PX4-Autopilot #7895: modules to consider retiring

dagar, 2017-09-01:
There are a few modules in examples (ekf_att_pos_estimator, mc_att_control_multiplatform, mc_pos_control_multiplatform) that are still in every build, changed somewhat regularly as uORB messages and the system evolves, but never actually tested. As a result it's quite likely at least some functionality within them has regressed.

If there's any value in keeping these around I think preserving their last known working state is more valuable than keeping them compiling, but possibly broken.

I propose retiring these modules either by moving them to a new external repository (setup as out-of-tree modules) or deleting them entirely.

The retired modules repository might be a good stepping stone for phasing out things like sdlog2.

tops4u, 2017-09-09:
Maybe if there is Space SF1xx Driver could be included in the standard build, as this takes little Space?

dagar, 2017-09-10:
@tops4u sf1xx is pretty small, we could readd it to px4fmu-v2_default.

tops4u, 2017-10-23:
@dagar is SF1xx going to be included in 1.7?

dagar, 2017-12-06:
Partially done as part of v1.7.0.

closed 2017-12-06 by dagar