PX4/PX4-Autopilot #7897: Stable firmware: Mission Protocol responding incorrectly over mavlink 1 protocol

DonLakeFlyer, 2017-09-01:
* Use a daily build of QGC which supports the new misison protocol. Turn on PlanManagerLog if you want to see the detaild protocol sequence.
* Use the current stable build of firmware (using default parameter set, hence mavlink 1)
* Try to upload a mission

You will get an upload failure. This is due to the firmware responding back with a MISSION_REQUEST with type=1 (should be type=0) during a mission upload sequence. Also note that in this case mavlink is running mavlink 1 protocol (since default parameter set). So QGC never asks for geofence or rally.

mwingerson, 2017-09-01:
In my experience, using the Pixhawk 2.1 with the PX4 1.6.5dev firmware with QGC V 3.2.4. I could not upload or download missions when the mavlink protocol selection was to force mavlink 2 or use mavlink1 and try mavlink 2 if GCS can support it.

When I forced mavlink 1 I had full functionality. 
Thanks Don!

DonLakeFlyer, 2017-09-02:
I need to think about the force mavlink 2 case which is not the case I was pointing out above. I'm not sure if there are still holes with the protocol in that case. I believe that case still has a problem with the latest changes Lorenz made for mavlink 2 auto-switch. Can't remember if he ever put the last fix through for that.

The case stated here is the default "use mavlink1 and try mavlink 2 if GCS can support it". While testing QGC since firmware is old, QGC does not switch to mavlink 2.

closed 2018-02-21 by PX4BuildBot