PX4/PX4-Autopilot #7899: MC: Add Lightware SF/LW 20 Support

tops4u, 2017-09-02:
**Feature Request**

Lightware is actually selling a very small and yet powerful (100m Range, 10gr) LIDAR for Altitude Measurements the SF20 (or LW20 that is in a Metall housing and weatherproof) that would be very convenient for UAV usage. It would be nice to have support for this in upcoming PX4 Releases. It is available as Serial and I2C Variant. I prefer the I2C Variant as Serial Interfaces are not easily sharable and scarce if some auxillaries are connected already. As there is I2C Support with the SF1xx Driver already this should not be too hard to implement.

dagar, 2017-09-02:
Do you have one to try? In the other variants the driver is identical, but you configure the min/max distance and rate.
https://github.com/PX4/Firmware/blob/master/src/drivers/sf0x/sf0x.cpp#L294

tops4u, 2017-09-02:
Yes I habe the device available and I would probably know how to öerform rhe changes but I don't have the Build Infrastructure and the default 1.6.5 Build lacks the SF 1xx Driver AFAIK.

tops4u, 2017-09-02:
Sorry for the Typos (mobile). I would want to use it with I2C. SF0x Driver is Serial.

dagar, 2017-09-02:
Do you mean on a pixhawk? If so you just need to add a line here: https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L30

Add an appropriate params here: https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensor_params.c#L1113

PR #7913 (2017-09-09): SF20/LW20 Support
Added basic support for SF20/LW20 I2C connected Lidar from Lightware. 

This would fix #7899
closed 2017-09-05 by tops4u