PX4/PX4-Autopilot #7910: data link loss failsafe do not work in take off state

jianyang1, 2017-09-04:
Data link failsafe action is set to RTL but its not executed in takeoff state.
that might potentially cause a dangerous situation of completely uncontrolled drone when using no RC at all.

can be produce in Gazebo using master. QGC 3.2.4 ,without joystic.

1. command Take Off
2. Drone climb to MIS_TAKEOFF_ALT
3. shut down QGC
4. after COM_DL_LOSS_T  "all data links lost"  message  is sent
5. bug here-  the drone keep loitering infinitely without any control

https://logs.px4.io/plot_app?log=80935714-f8a4-45b9-8881-bfef1a3225f6

regardless, i think drones flight mode should be changed automatically to Hold after it's reached takeoff alt, as well as when finishing mission for example

BazookaJoe1900, 2017-09-05:
Looking on the set_nav_state() function that transit the states i might see other potential failures.
for example, engine_failure and gps_failure  on offboard, are not being handled.

dagar, 2017-10-11:
PR to switch automatically switch to HOLD (or MISSION) after TAKEOFF is completed successfully.

closed 2018-02-21 by PX4BuildBot